aredn/aredn #633: AP doesn't boot with active WAN

KenArck, 2023-01-10:
Nightly 2124-6dac4c6 

hAP AC2 - if booted (cold or warm) with WAN connection made, AP doesn't broadcast SSID so can't connect. If booted  (cold or warm) without WAN connection, AP comes alive
[supportdata-AH6LE-HAP-SHOP-202210191622.gz](https://github.com/aredn/aredn/files/10377901/supportdata-AH6LE-HAP-SHOP-202210191622.gz)
 comes alive.

aanon4, 2023-01-10:
Thanks. Did this work okay on a previous nightly (it's probably a firewall change)?

KenArck, 2023-01-10:
It worked as of 3.22.12 (the last version I flashed to my hAP Lite). It only started happening as of the first nightly build made available for the hAP AC2 ( I believe I first placed it into service on Monday, Jan 3rd). I've since reflashed with build I posted in my OP

aanon4, 2023-01-11:
Can you describe what's plugged into you hAP AC2? Is there just the single cable plugged into the WAN port? If not, what else is connected. Do these cables plug into switches? Are any external VLANs configured?
Thanks.

KenArck, 2023-01-11:
WAN port is plugged directly into TPLink AX1800 Router. Port 1 connected to TPLink TL-SG108E switch. Port 2 connected to GrandStream GXP1620 phone

aanon4, 2023-01-11:
Could I request the following while running the current nightly:
1. Boot without WAN - wait 10 minutes - check the status of the AP - send support data.
2. Boot with WAN - wait 10 minutes - check the status of the AP - send support data.
Thanks!

KenArck, 2023-01-11:
Only problem is it will not boot with WAN hence I can't download the support file

aanon4, 2023-01-11:
I have my dev hAP AC2 on my desk (it has a serial cable connected which is helpful at times). If I boot it up with just a WAN cable connected it does boot. For me when I insert the power, the PWR light comes on, then in a moment the USR light comes on indicating the device is in the boot loader. After a while this light goes off again, then starts to flash as the AREDN software boots. Eventually it goes steady again as boot is completed.  Very occasionally I've noticed inserting the power never has the USR light come and the device doesn't correct power up (no output on the serial line). This hasn't seemed correlated to what cables I have inserted. I've found unplugging for a dozen seconds the retrying helped (immediately retrying didn't).

Anyway, booting up with WAN only cable at the moment and it's currently sitting not finding my -2 mesh network, but that can take 3-10 minutes sometimes on this device (another bug, but at least only on boot).

KenArck, 2023-01-11:
I've waited longer than 15 minutes for it to boot with WAN and it never did. I have noticed the power LED as you mentioned - it does come on solidly as you describe but no boot. 

What's this serial connection you're referring to?

KenArck, 2023-01-16:
Just loaded the brand new nightly build (2166-fdeda7d ) and no improvement to the WAN problem as reported in my OP

aanon4, 2023-01-16:
Re-reading these notes I'm confused about one thing. At one point you mention the hAP Lite but mostly the hAP AC2. How did the Lite impact this?

Also, can you try booting up the hAP AC2 without the WAN connected, then plug it in once you see the USR light flashing?

It's not no help to you I realize, but have an hAP AC2 running as my tunnel server, and I know of a couple more that have been deployed.

KenArck, 2023-01-16:
I replaced my hAP Lite with the AC2 (the Lite never had this problem) when I saw there was now firmware for the AC2 available. 

As for the WAN connection, yes. If I remove the WAN connection while the AC2 is booting and plug the WAN cable back in after I see the access point is broadcasting, all is good.

aanon4, 2023-01-16:
Odd. Although I had exactly your problem but with the hAP Lite - with the WAN plugged in it would often fail to reboot, or take forever to do it.

aanon4, 2023-01-16:
If you plug the WAN port into something else will it boot up - say your laptop LAN port?

aanon4, 2023-01-16:
Did a little rummaging around and found this discussion:
https://forum.openwrt.org/t/openwrt-installed-mikrotik-rb750gr3-wont-boot-when-wan-cable-plugged/146330/10
Not this device, but possibly related. Will look into this.

aanon4, 2023-01-16:
Can you log into the problem node and type the following:
```
echo "flasheth" > /sys/firmware/mikrotik/soft_config/boot_device
echo "1" > /sys/firmware/mikrotik/soft_config/commit
```
Then reboot with the WAN connected and see if the problem is resolved.

aanon4, 2023-01-16:
I've created a PR to do this automatically once you confirm it fixes the issue.

KenArck, 2023-01-16:
Yes, I ran those commands and not only did it reboot fine with the WAN active, it seems to reboot much faster than previously

closed 2023-01-17 by aanon4