aredn/aredn #636: Issues w/ hAP AC3 Wireless

km6lslf, 2023-01-11:
I've noticed a couple issues with the hAP AC3. They were in place on the last nightly I installed from a few days ago, and remain issues in 2138-2f02c87. 
When booted up in close proximity to nodes it should connect to, it doesn't make any connections.
[image]
Also, despite having WiFi client enabled and SSID and password information saved, it does not connect to nearby WiFi on reboot. 
I noticed the system time was off by about 25 minutes, so I changed the NTP sync setting to hourly. No change after reboot. Support data was taken prior to the following steps, but I get an error when I try to attach it. 

Running a WiFi Scan seemed to "wake up" the mesh RF. 
[image]

aanon4, 2023-01-11:
There seem to be several separate issues here - can you split them into separate issues so we can address and track properly? I'm going to keep this one *only* for the Wifi issue.

Regarding the Wifi issue - please wait at least 5 minutes and see if the nodes connect. There are problems here for sure, but usually the node will recover by the 5 minute mark. If after 5 minutes it hasn't, please wait another 5 then snd over the support data - thanks.

km6lslf, 2023-01-11:
Will do. Here's an update on the WiFi issue:

I rebooted a couple times with different settings to see if anything would get the node to connect to my home WiFi. After some trial and error, it looks like it's connected. 
[image]
I tried connecting using the 5GHz radio, but no connection was established. Trying again, setting the WiFi client band to 2.4GHz, it was able to connect. 
[image]

Could the problem be with the 5GHz radio in general? I tested connecting to it with the node in WiFi AP mode, and it worked.

Connecting to the WiFi has fixed the device time issue. With that fixed, I'm trying turning mesh RF back on and setting the 5GHz radio to client.

km6lslf, 2023-01-11:
The device apparently automatically connected to nearby nodes after a short period of time, but the WiFi still doesn't work. I'll hold off on opening a separate issue since the mesh RF might not be a problem after all.

wck86, 2023-01-12:
Tim, further investigation yielded some interesting results. WiFi signals were being detected, just not the one I was trying to connect to, for some reason. My primary home WiFi's 5GHz signal is 40 MHz wide, and I noticed that none of the detected SSIDs had 40 MHz wide channel widths. I'll try changing to 80MHz and will report back if that gets it to connect.

root@KM6SLF-hAPac3-80-200:~# iw dev wlan1 scan
BSS 58:d9:d5:73:ce:25(on wlan1)
        last seen: 22901.290s [boottime]
        TSF: 43683333612 usec (0d, 12:08:03)
        freq: 5180
        beacon interval: 100 TUs
        capability: ESS (0x0511)
        signal: -95.00 dBm
        last seen: 3990 ms ago
        Information elements from Probe Response frame:
        SSID: C69DFB_5GEXT
        HT capabilities:
                Capabilities: 0x11ef
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 32767 bytes (exponent: 0x002)
                Minimum RX AMPDU time spacing: 2 usec (0x04)
                HT RX MCS rate indexes supported: 0-15
                HT TX MCS rate indexes are undefined
        HT operation:
                 * primary channel: 36
                 * secondary channel offset: above
                 * STA channel width: any
        VHT capabilities:
                VHT Capabilities (0x03c161b0):
                        Max MPDU length: 3895
                        Supported Channel Width: neither 160 nor 80+80
                        RX LDPC
                        short GI (80 MHz)
                        TX STBC
                        +HTC-VHT
                VHT RX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: not supported
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT RX highest supported: 780 Mbps
                VHT TX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: not supported
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT TX highest supported: 780 Mbps
        VHT operation:
                 * channel width: 1 (80 MHz)
                 * center freq segment 1: 42
                 * center freq segment 2: 0
                 * VHT basic MCS set: 0xfff0
        WPA:     * Version: 1
                 * Group cipher: CCMP
                 * Pairwise ciphers: CCMP
                 * Authentication suites: PSK
        RSN:     * Version: 1
[... 315 lines cut ...]
                VHT RX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: not supported
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT RX highest supported: 0 Mbps
                VHT TX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: not supported
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT TX highest supported: 0 Mbps
        VHT operation:
                 * channel width: 1 (80 MHz)
                 * center freq segment 1: 122
                 * center freq segment 2: 0
                 * VHT basic MCS set: 0xfffc

wck86, 2023-01-12:
After switching my home WiFi (Ubiquiti Dream Machine) 5GHz to 80 MHz wide the hAP AC3 was able to connect to it. It looks like the issue is specific to 40 MHz wide channels.