aredn/aredn #637: Package Management shows ~40 options to download 'Package version' in 2131 build

ae6xe, 2023-01-11:
on Mikrotik ac2, after refreshing the 'download packages', there were ~40 selection entries to download 'package version'.  The actual package names are not showing, except 'kmod*' and 'block-mount*'.   I was looking for tcpdump to help debug some issues.  On 2131 nightly build image.  The refresh occurred before the next nightly build.

dman776, 2023-01-25:
may be specfic to the ipq40xx architecture.    It works find on my hAP AC Lite (generic/mikrotik)

dman776, 2023-01-25:
what are the values in your AdvConfig firmware paths?   are they correct for the arm_cortex..... packages?

dman776, 2023-01-25:
it’s hard coded in the advandedconfig page in the defaultPackageRepos() function around line 77
need to pull that value from “/etc/openwrt_release DISTRIB_ARCH” instead

ae6xe, 2023-01-25:
Confirmed, all the paths on the hap ac2 need to have changed in the download paths:  from "mips_24kc", to "arm_cortex-a7_neon-vfpv4".

dman776, 2023-01-25:
fixed by #672

closed 2023-01-25 by dman776