aredn/aredn #640: Mikrotik LDF5 Not upgrading

kc2cbd, 2023-01-12:
Tried to use the new firmware for the LDF5 and it doesn't upgrade

`for i in /tmp/sysinfo/* ; do echo $i ; cat $i; done`
produces
`root@KC2CBD-LDF5-151-CKV-50TWR-106:~# for i in /tmp/sysinfo/* ; do echo $i ; cat $i; done
/tmp/sysinfo/board_name
rb-ldf-5nd
/tmp/sysinfo/boardid
MikroTik RouterBOARD LDF-5nD
/tmp/sysinfo/hardware_mfg
MikroTik
/tmp/sysinfo/model
MikroTik RouterBOARD LDF-5nD
`

Is this going to be another that will have to be upgraded as if it's a new device and than it will work from there?

aanon4, 2023-01-12:
https://github.com/aredn/aredn/pull/639

kc2cbd, 2023-01-13:
Still not upgrading from Firmware 2002

aanon4, 2023-01-13:
The PR has not been merged yet.

kc2cbd, 2023-01-14:
Upgrade successful and appears to be functioning correctly,

closed 2023-01-14 by kc2cbd