aredn/aredn #643: Invalid Signature File for packages?

wck86, 2023-01-12:
IIRC, I believe this was not an issue during a prior nightly release, but under the one from 1/11 for the hAP AC3.

Hitting Refresh on the Administration screen, next to the Download Packages drop-down triggers the attached results:
<html>
<body>

Downloading http://downloads.arednmesh.org/snapshots/targets/ipq40xx/mikrotik/packages/Packa ges.gz Updated list of available packages in /var/opkg-lists/aredn_core Downloading http://downloads.arednmesh.org/snapshots/targets/ipq40xx/mikrotik/packages/Packa ges.sig Signature check failed. Remove wrong Signature file. Downloading http://downloads.arednmesh.org/snapshots/packages/mips_24kc/base/Packages.gz Updated list of available packages in /var/opkg-lists/aredn_base Downloading http://downloads.arednmesh.org/snapshots/packages/mips_24kc/base/Packages.sig Signature check failed. Remove wrong Signature file. Downloading http://downloads.arednmesh.org/snapshots/packages/mips_24kc/arednpackages/Packag es.gz Updated list of available packages in /var/opkg-lists/aredn_arednpackages  Downloading http://downloads.arednmesh.org/snapshots/packages/mips_24kc/arednpackages/Packag es.sig Signature check failed. Remove wrong Signature file. Downloading http://downloads.arednmesh.org/snapshots/packages/mips_24kc/luci/Packages.gz Updated list of available packages in /var/opkg-lists/aredn_luci Downloading http://downloads.arednmesh.org/snapshots/packages/mips_24kc/luci/Packages.sig Signature check failed. Remove wrong Signature file. Downloading http://downloads.arednmesh.org/snapshots/packages/mips_24kc/packages/Packages.gz  Updated list of available packages in /var/opkg-lists/aredn_packages Downloading http://downloads.arednmesh.org/snapshots/packages/mips_24kc/packages/Packages.si g Signature check failed. Remove wrong Signature file. Downloading http://downloads.arednmesh.org/snapshots/packages/mips_24kc/routing/Packages.gz  Updated list of available packages in /var/opkg-lists/aredn_routing Downloading  http://downloads.arednmesh.org/snapshots/packages/mips_24kc/routing/Packages.sig  Signature check failed. Remove wrong Signature file. Downloading http://downloads.arednmesh.org/snapshots/packages/mips_24kc/telephony/Packages.g z Updated list of available packages in /var/opkg-lists/aredn_telephony Downloading http://downloads.arednmesh.org/snapshots/packages/mips_24kc/telephony/Packages.s ig Signature check failed. Remove wrong Signature file.
--

</body>
</html>

[image]

aanon4, 2023-01-12:
This means you're not running the current nightly build.

dman776, 2023-01-12:
You need to be on the new nightly build.   One was generated AFTER your install

closed 2023-01-12 by dman776

wck86, 2023-01-13:
A day late and a buck short, thanks guys!