aredn/aredn #645: GL-iNet USB150 bricked when updated to 2145 or 2152

k6ccc, 2023-01-13:
Updated all 14 of my nodes to Nightly 2145 and all worked except my USB150.  The 2145 update was made over an RF link of about 6 feet.  When I got home and checked it out, a computer would recognize that a USB device was plugged in, but could not get an IP address from it.  Power cycled several times with the same result.  I don't remember for sure what the previous version was, but it was a nightly from sometime in the past week.

This morning used the GL-iNet unbrick process to unbrick the USB150.  Then did an update to 3.22.12.0  That properly came up with the first boot screen and then after the next reboot, it was operating properly as far as I could tell.  I don't have any other 2 GHz AREDN here to test an AREDN RF link with.  Next updated to nightly 2152.  Again, the USB150 bricked.  The attached computer recognizes that a USB device has been plugged in, but can not get an IP address from it.

Repeated the unbrick and then AREDN load and update process with the same result.

aanon4, 2023-01-13:
I'll dig out my USB150 this weekend and take a look. My guess is that something changed specifically for this device when we did the OpenWRT upgrade to their latest version earlier this week. Anyway, dont try an upgrade on the USB150 again until I update this ticket.

k6ccc, 2023-01-13:
I'm gonna leave it at 3.22.12.0 for now...

kc2cbd, 2023-01-13:
The last couple of version of nightly have been weird with the USB150. It now sort of powers down and doesn't come back up after upgrade.

k6ccc, 2023-01-14:
Update.  KC2CBD hit the nail on the head.  After the update, the node powers off and does not power back on.  However after updating, if you wait for the node to power off (about 4 minutes), then you power cycle it, and wait several more minutes, it will come back up with the new firmware and appears to be working correctly.

kc2cbd, 2023-01-14:
I was actually going to create an issue but K6CCC beat me to it.

k6ccc, 2023-01-14:
Same issue with nightly 2159.  After the update the USB150 powers down, but does not power back up.  Must be manually power cycled to get it to boot up.

k6ccc, 2023-01-15:
Still the same thing with NB 2166.  Node powers down, but does not power back up after firmware update.  Requires manual power cycle.

aanon4, 2023-01-15:
Thanks for checking. I promise to update this ticket when there's any relevant changes though ... if you want to save yourself some time.

aanon4, 2023-01-16:
Update:
Looks like the device fails while rebooting. Unfortunately the serial line is not configured at the point it crashes and is just printing junk ... but given the amount of junk it's printing it's gotten past the boot loader and has started running the kernel, but fails and halts. Why? I dont know yet, and I probably wont until I get some real serial output.
I do *not* expect any nightly to fix this as there is probably something broken in the kernel of this device right now.

kc2cbd, 2023-01-17:
2180 was new..it took the firmware and then basically shut off. I then power cycled and like previously it was updated to 2180.

kc2cbd, 2023-01-19:
2187 seems to have fixed the USB150 also.

k6ccc, 2023-01-20:
Neither 2187 nor 2195 fixed this for me.  Both required multiple power cycles to bring the USB150 back onto the mesh.

aanon4, 2023-01-20:
Ever optimistic .. but I dont have anything in the works to address this problem ... hoping for some revelation at some point.

k6ccc, 2023-01-20:
I moved the "storage" location for the USB150 at home to an outlet that is remotely controllable, so easy to power cycle now...

dman776, 2023-01-25:
@aanon4  here is an openwrt issue that looks like the same issue (with the fix) ...  https://forum.openwrt.org/t/reboot-command-fail/113250/3

aanon4, 2023-01-26:
Unfortunately, adding *broken-flash-reset* to the DTS did not fix things.

aanon4, 2023-01-26:
Also, historically, the serial port was disabled - but I tried disabling that again (in case there was some side-effect in doing that) but that also made no difference.

aanon4, 2023-01-26:
Flashing the OpenWRT reference image shows the same behavior so .. in case there was any doubt .. this problem is not related to our configuration or changes.

kc2cbd, 2023-01-31:
The last couple of firmware releases have been working correctly.

k6ccc, 2023-01-31:
Concur.  2245 did not require a power cycle, and I just updated to 2267 and that did not require a power cycle on the USB150.

closed 2023-01-31 by dman776