aredn/aredn #648: NB 2159 - Nodes that have DtD links are reporting some RF links as Blocked - dtd

k6ccc, 2023-01-15:
I control 14 nodes that have a collection of RF and three separate DtD connections.  All 14 nodes were updated to NB 2159 today.
A few hours after the updates, I discovered a node showing on the Neighbor Status page a RF link wth a Status of blocked - dtd.  Started checking other nodes, and every single node that has one or more DtD neighbors AND one or more RF links is reporting some of the RF links as blocked - dtd.  Nodes that are RF only do not show this issue.
It appeared that I could NOT communicate over the affected RF links (if there was no other path).

k6ccc, 2023-01-15:
Support data for a Rocket M5 and a hAP showing this issue.  More available if desired.

[supportdata-K6CCC-ROCM5-1-GLENDALE-202301142008.gz](https://github.com/aredn/aredn/files/10419421/supportdata-K6CCC-ROCM5-1-GLENDALE-202301142008.gz)
[supportdata-K6CCC-hAP-at-Home-202301142009.gz](https://github.com/aredn/aredn/files/10419422/supportdata-K6CCC-hAP-at-Home-202301142009.gz)

k6ccc, 2023-01-15:
On a small sample size, reboot (or several) appears to clear this.  Unknown yet if the fix lasts...
This support data for one of the above nodes a while AFTER two reboots.

[supportdata-K6CCC-ROCM5-1-GLENDALE-202301142035.gz](https://github.com/aredn/aredn/files/10419454/supportdata-K6CCC-ROCM5-1-GLENDALE-202301142035.gz)

aanon4, 2023-01-15:
https://github.com/aredn/aredn/pull/649

aanon4, 2023-01-15:
Bit of a stab in the dark if I'm honest as I can't see how this could happen with the current code ... but various people have reported the problem. I've modified to code to be more active when choosing the block and unblocking by default - we'll see if that improves things.

k6ccc, 2023-01-15:
Glad I was not the only one seeing this one...

aanon4, 2023-01-17:
Has this been fixed in the latest nightlies?

ab7pa, 2023-01-17:
Yes, the nightly build code has fixed this issue for me on the SXTsq5ac.  THX

k6ccc, 2023-01-17:
Yes, it has been fixed on the various nodes where I observed the problem.

closed 2023-01-17 by dman776