aredn/aredn #657: device version file naming is confusing (XM vs XW, WA vs XC)

cTurtle98, 2023-01-18:
let me know if this is just me

I find it very confusing to have our builds such that one device file is named with just the device name and one is named device name - model

it feels not very explicit and makes ctrl+f searching the page more dificult

could we have it so they all have the model at the end?

no
`nanobeam-ac` and `nanobeam-ac-xc`

yes
`nanobeam-ac-wa` and `nanobeam-ac-xc`

maybe I'm alone in this but I thought I would test the waters

aanon4, 2023-01-18:
The build names are largely dictated by OpenWRT. We've changed them in the past but it just breaks a lot of the tooling, so we just go with their format.

dman776, 2023-01-18:
just use the AREDN Firmware Selector (AFS) tool and you wont need to even know the names...
http://downloads.arednmesh.org/afs/www

closed 2023-01-20 by aanon4