aredn/aredn #662: GL iNet USB150 allowing tunnel via the mesh when should be allowed with NB2195

k6ccc, 2023-01-20:
Had my USB150 on NB 2187 running with the RF being used for the mesh - so no WAN connection.  Two tunnels were enabled on the Tunnel Client page, but they were not operating because of no WAN - so far so good.
Updated to NB 2195, and after the two required power cycles to get the node back up at all, the RF path connected, but then I noticed that both tunnels were operating.  Verified that WAN-Only tunnel is enabled on the Advanced Config page.  Disabled the tunnels on the Tunnel Client page and tunnels went away as expected.  Re-enabling either or both tunnels did NOT bring the tunnels back up.  Therefore, this may be VERY difficult to reproduce.
The screen capture and support file were taken while the tunnels were operating before disabling them.
[image]
[supportdata-K6CCC-USB150-Portable-202301200838.gz](https://github.com/aredn/aredn/files/10469768/supportdata-K6CCC-USB150-Portable-202301200838.gz)

ab7pa, 2023-02-08:
@k6ccc Jim, is this still an issue?  My usb150 is on NB2293 now but I have not been able to reproduce these symptoms.

k6ccc, 2023-02-08:
My USB150 is also now on NB2283 and I enabled the tunnels and it properly did not build the tunnels over the mesh.

aanon4, 2023-02-08:
Seems we're good to close then. Please reopen if necessary.

closed 2023-02-08 by aanon4