aredn/aredn #674: Nightly 2224 (and previous) limit TX Power to 25 dBm

nc8q, 2023-01-25:
sysinfo:
 node: NC8Q-Kettering-Centerville
model: MikroTik RouterBOARD LHG 5nD
status:
firmware version: 2224-4894f42
model: MikroTik RouterBOARD LHG 5nD
This is actually a MikroTik RouterBOARD LHG 5HPnD-XL and is being limited to 25 dBm.

sysinfo:
MikroTik RouterBOARD LHG 5HPnD-XL
 node: NC8Q-Centerville-Kettering
model: MikroTik RouterBOARD LHG 5HPnD-XL
status:
firmware version: 3.22.12.0
model: This is correct and allows 28 dBm

sysinfo:
 node: NC8Q-Centerville-HuberHeights
model: MikroTik RouterBOARD RBLHG-5HPnDstatus:
status:
firmware version: 3.22.12.0
model: MikroTik RouterBOARD RBLHG-5HPnD
This is correct and allows 28 dBm

sysinfo:
 node: NC8Q-HuberHeights-Centerville
model: MikroTik RouterBOARD RBLHG-5HPnDstatus:
firmware version: 1827-2711589
model: MikroTik RouterBOARD RBLHG-5HPnD
This is correct and allows 28 dBm

vy 73, Chuck

ae6xe, 2023-01-25:
The max power settings have some constraints to consider.   As an example, on my hap ac2, the linux driver will allow a setting up to 30dBm on all channels (AREDN settings may further restrict max power).    If we refer back to the vendor specifications, the max is 27dBm on the hap ac2, and higher link rates are lower power.   See attached spec for hap ac2 vendor specs.
[image]

AREDN should limit the hap ac2 to 27dBm max, otherwise the signal is at risk of being distorted.  Generally this is due to cheap power amps, and avoidance to over-drive (out of linearity) this PA and bad things happen -- noise outside the channel, distorted signal, etc.   Higher link rates (vs lower link rates) can also cause the PA to be over-driven, hence the vendor specs and firmware limiting the max power further. 

We should set the max power to be from the vendor specs.   If AREDN allows a higher power settings, there is risk of a poor quality signal and possibly making a worse quality and throughput link.   A spectrum analyzer would confirm.     Then, we could add documentation for the node admin's to test lower power settings if the link, particularly if the node is reaching higher link rates -- can power be reduced and still keep or exceed through-put rates?  

It's counter intuitive... possibly many devices deployed today with max power settings at MCS15 rates, have some signal distortion and a slightly lower power may yield higher through-put.

nc8q, 2023-01-25:
Hi, Joe:
The SXTsqHP5 and LHG-HP5 are the same radio, I think.
The same 3.22.12.0 firmware matches LHG/SXTsq/5/HP5 and
whatever maximum power is specified is configurable.
The Nightlies permit 28dBm in the SXT, so they should also permit LHG-HP5s 28 dBm.
73, Chuck
[image]

aanon4, 2023-01-25:
I need to split out the LHG variants into their own builds (which is what I really should have done from the beginning) which will fix this issue.

ae6xe, 2023-01-25:
The LHG has a high power and a low power variant.   Indeed the vendor specs for the low power variant is max 25dBm:  https://mikrotik.com/product/RBLHG-5nD.      The RBLHG-5HPnD specs are 28dBm max.  The motherboard may be the same from a linux and driver perspective, but the hardware (particularly the power amp) appears not to be on these 2 variants.

On the low power variant, do the command "iw phy phy0 info" and see what the max power shows by each channel (this is defined in the vender flash partition).   Even if AREDN attempts to exceed this value, the radio will not go higher than what is shown by this iw command.

aanon4, 2023-01-27:
Can we close this?

nc8q, 2023-01-27:
Yes.

closed 2023-01-27 by nc8q