aredn/aredn #680: Nightly build 2238-ee3fa7f doesn't support/detect LHG 2HPnD-XL properly

bwarden, 2023-01-28:
I had to force sysupgrade to install this on my LHG 2nD-XL:
http://downloads.arednmesh.org/snapshots/targets/ath79/mikrotik/aredn-2238-ee3fa7f-ath79-mikrotik-mikrotik_routerboard-lhg-2hpnd-xl-squashfs-sysupgrade.bin

Afterward, I get the dreaded *UNSUPPORTED DEVICE* banner.
sysinfo:
```
node: N7BTW-QTH-A
model: MikroTik RouterBOARD LHG 2HPnD-XL

!!!! UNSUPPORTED DEVICE !!!!
boardid: MikroTik RouterBOARD LHG 2HPnD-XL
Device HAS BEEN TESTED AS UNSUPPORTED

br-dtdlink 02:E1:8D:55:63:CC
br-lan 08:55:31:23:B5:07
br-wan 02:EA:60:8B:E6:6C
br0    1E:53:54:B6:14:8F
br0.1  1E:53:54:B6:14:8F
br0.2  1E:53:54:B6:14:8F
br0.3  1E:53:54:B6:14:8F
eth0   1E:53:54:B6:14:8F
tun50  00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
tun60  00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
tunl0  00-00-00-00-FF-FC-00-44-00-00-00-00-00-00-00-00
wlan0  08:55:31:23:B5:08
wlan0-1 0A-55-31-23-B5-08-00-44-00-00-00-00-00-00-00-00

/proc/cpuinfo
system type		: Qualcomm Atheros QCA9533 ver 2 rev 0
machine			: MikroTik RouterBOARD LHG 2HPnD-XL
processor		: 0
cpu model		: MIPS 24Kc V7.4
BogoMIPS		: 432.53
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 16
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16
Options implemented	: tlb 4kex 4k_cache prefetch mcheck ejtag llsc dc_aliases perf_cntr_intr_bit perf
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

nvram
hsmmmesh.settings=settings
hsmmmesh.settings.wifimac='08:55:31:23:b5:08'
hsmmmesh.settings.mac2='35.181.8'
hsmmmesh.settings.dtdmac='35.181.7'
hsmmmesh.settings.node='N7BTW-QTH-A'
hsmmmesh.settings.config='mesh'
hsmmmesh.settings.nodeupgraded='0'

Part of the AREDN™ Project. For more details please [see here](http://localhost:9001/about.html)
```

Looks like we're just missing the match (MikroTik RouterBOARD LHG 2**HP**nD-XL) in `/etc/radios.json`:
```
# grep -i LHG.*2nD.*XL /etc/radios.json
    "MikroTik RouterBOARD LHG 2nD-XL" : {
      "name"            : "MikroTik RouterBOARD LHG 2nD-XL",
    "MikroTik RouterBOARD RBLHG 2nD-XL" : {
      "name"            : "MikroTik RouterBOARD RBLHG 2nD-XL",
    "MikroTik RouterBOARD RBLHG-2nD-XL" : {
      "name"            : "MikroTik RouterBOARD RBLHG-2nD-XL",
```

aanon4, 2023-01-28:
It looks like I made a couple of mistake here and that there is no 'hp' model of the LHG 2, just the base and an XL version. I'll fix that.

aanon4, 2023-01-28:
https://github.com/aredn/aredn/pull/682

aanon4, 2023-02-05:
Fixed?

bwarden, 2023-02-06:
> Fixed?

Yes, thanks!

closed 2023-02-06 by aanon4