aredn/aredn #686: UBNT Airrouter not fully functional after the new OpenWRT Nightly releases since N-2002

k1ky, 2023-01-29:
UBNT AirRouter "upgrades" but the first GUI screen doesn't fully load, nor does the MESH RF seem to function - or at least other nodes don't see it.  All other screens work, indicating that the new firmware loaded with the proper nightly upgrade version that was installed. .  Unit comes back at 192.168.1.1 and then 10. addy after 2nd reboot.  Am able to "downgrade" back to N-2002, which restores the unit to proper operation. Unable to get a support file.

aanon4, 2023-01-29:
If possible could you log into the node, refresh the failing status page, then type ```logread``` and post the output here - particularly the last dozen or so lines which will (hopefully) contains some errors.

k1ky, 2023-01-29:
Here is the screen:
[image]
It's accessible at 192.168.1.1
Working on the logread - it's a very long report.

k1ky, 2023-01-29:
Let's try this file
[logread.txt](https://github.com/aredn/aredn/files/10529111/logread.txt)

aanon4, 2023-01-29:
https://github.com/aredn/aredn/pull/687

k1ky, 2023-01-29:
Failure to save setup when attempting to change channel - Also asked to reset password:
[image]

aanon4, 2023-01-29:
Dont bother testing anything else until that first page works - I submitted a fix

k1ky, 2023-01-29:
Copy that.