aredn/aredn #694: UBNT AirRouter doesn't allow Wi-Fi AP or WAN Nightly tiny 2274

k1ky, 2023-02-03:
Running Nightly 2274 doesn't have the RF Wi-Fi AP nor WAN selections available on units previuosly configured for such operation.
[image]

aanon4, 2023-02-03:
Tiny builds no longer include support for AP access by default due to the lack of space. The packages can be installed separately if you're in a small enough mesh environment where this wont cause problems. I'll make a note in the README.

k1ky, 2023-02-03:
OK - Kinda figured that.  Can you point me to which module(s) I need to load in order to restore this function? At least the Wi-Fi AP mode as it is driving our Weather station at one of my repeater sites.

aanon4, 2023-02-03:
Look for "wpad-mini" ... not the most obvious name.

k1ky, 2023-02-03:
The wpad-mini install did the trick.  It also retained my previous AP setting and my Weather Station has reconnected and is reporting properly. On another note, I was able to "revert" another AirRouter tunning N-2074 back to N-2002, which restored the AP capability running the older Nightly firmware.  Thanks for the response.  Should I wait to close this until you update the README file?  This should be a "warning" to folks upgrading to the new OpenWRT tiny nightlies when loss of funtion is involved.

aanon4, 2023-02-03:
Let's wait until this PR is merged:
https://github.com/aredn/aredn/pull/695

k1ky, 2023-02-03:
Please be as specific as possible on critical module functions lost.  I'm aware of Tunnel and now the Wi-Fi AP and WAN functions.

aanon4, 2023-02-03:
I'll talk to Steve about adding something to the docs about this - dont want to start listing all the potential packages for various scenarios here.

ab7pa, 2023-02-03:
That information goes into the **Release Notes**.  The online docs do not contain any information about specific firmware builds or packages -- all of that information is always maintained in the _Release Notes_.

ae6xe, 2023-02-03:
We document the port usage per device in the github README.    Wherever we
put device specific information, we should try to keep it all together.
 Note, device specific details are relevant across many releases.     It
doesn't seem to be the best place to put in the notes of a specific
release.  Is github the best place?

k1ky, 2023-02-12:
Ready to close this one?

aanon4, 2023-02-12:
Yes. I've moved copies of the release notes we post online into the git so people can easily refer to them here. We'll add appropriate notes when we do the final release.

closed 2023-02-12 by aanon4