aredn/aredn #699: Migrate OpenWRT hwmode to new band parameter

ab7pa, 2023-02-08:
Evidently a couple of years ago OpenWRT deprecated the legacy `hwmode` parameter in favor of the new and more intuitive `band` parameter.  This may be a good thing for our firmware to migrate to as well.

[image]

Orv, 2023-02-09:
Is the 3 GHz band supported with band?

ae6xe, 2023-02-09:
3GHz devices are 5GHz motherboards, with -2GHz down converters.   Linux and the Qualcom wifi chip only have awareness of 5GHz.   Thus, the band is 5g. We have to detect in the model or board ID that it is a 3GHz device and set UI info accordingly.

Same for 900MHz devices, a 2GHz motherboard with a downconverter.

aanon4, 2024-10-14:
https://github.com/aredn/aredn/pull/1630

closed 2024-10-14 by aanon4