aredn/aredn #711: hAP lite reporting nodes as hidden when they aren't - NB 2283 and a few earlier.

k6ccc, 2023-02-12:
At home, I have four nodes that are all able to communicate with each other on 2.4 GHz (channel -2).  All four nodes are within six feet of each other, and can obviously see each other just fine.  Until recently, all four nodes would report direct connections on the Mesh Status and Neighbor Status pages.  The four nodes are:
K6CCC-hAP-at-Home   (a hAP ac lite)
K6CCC-AR750-Portable
K6CCC-RB-LHG-2nD-XL   (just the driven element - no reflector)
K6CCC-USB150-Portable

For the past few nightly builds, All of the nodes are reporting that they directly connect to each other except the hAP.  The hAP will report one of the other nodes as a neighbor, but the other two as hidden.  Which node is connected changes regularly.  When this screen capture was taken, the hAP was connected to the USB150, but that changes regularly.  I did a traceroute from the hAP and it also confirms that the hAP believes that the other two nodes are hidden.

Breaking news:  Updated all four nodes to NB 2309 and after waiting for all four nodes to change Neighbor status from Pending, they are all reporting that they are directly connected.  Traceroute on the hAP confirms direct connection.  So this may be fixed in 2309...

[image]
[hAP TraceRoute.txt](https://github.com/aredn/aredn/files/10716793/hAP.TraceRoute.txt)
[supportdata-K6CCC-hAP-at-Home-202302120934.gz](https://github.com/aredn/aredn/files/10716794/supportdata-K6CCC-hAP-at-Home-202302120934.gz)
[supportdata-K6CCC-AR750-Portable-202302120934.gz](https://github.com/aredn/aredn/files/10716795/supportdata-K6CCC-AR750-Portable-202302120934.gz)
[supportdata-K6CCC-RB-LHG-2nD-XL-202302121735.gz](https://github.com/aredn/aredn/files/10716796/supportdata-K6CCC-RB-LHG-2nD-XL-202302121735.gz)
[supportdata-K6CCC-USB150-Portable-202302120936.gz](https://github.com/aredn/aredn/files/10716797/supportdata-K6CCC-USB150-Portable-202302120936.gz)

aanon4, 2023-02-12:
Going to hold off investigating this unless the issue returns.

aanon4, 2023-02-20:
Closing as there's been no update on this with current firmware.

closed 2023-02-20 by aanon4