aredn/aredn #720: NB 2339 on hAP removes AP band selection if Mesh RF is on - 2 GHz only

k6ccc, 2023-02-18:
On Nightly build 2339 on a hAP lite, if Mesh RF is enabled on a node that has the WiFi AP enabled on 5 GHz, the band selection for the AP is removed and the channels are forced to 2 GHz.  Then the node complains that you can't have Mesh RF and LAN AP on the same wireless card.
This screen capture is before I made any changes:
[image]

After enabling Mesh RF, AP Band selection i not longer there and only 2 GHz channels available for selection.
[image]

After hitting Save, it bitches about Mesh and LAN AP trying to use the same wireless.
[image]

After disabling Mesh RF and saving, the Band selection for the LAN AP returns.

k6ccc, 2023-02-18:
Similar problem on GL iNet AR750.  This node prior to upgrade to 2339 had the LAN AP enabled on 5 GHz and Mesh RF enabled on 2.4 GHz.  After updating to 2339, this is what I see on the basic setup page.  According to a WiFi analyzer on my phone, the AR750 really is operating the AP on channel 44 on 5 GHz - not channel 1 as indicated.
[image]

k1ky, 2023-02-19:
It appears to work ok if MESH RF is NOT enabled, but otherwise pretty messed up when MESH RF is enabled.

aanon4, 2023-02-20:
https://github.com/aredn/aredn/pull/724

k6ccc, 2023-02-20:
Looks like it is fixed on NB 2347 - tested on both hAP lite and iNet AR750.

k6ccc, 2023-02-20:
Well, not so fast.  I was able to update my newer hAP ac3 to NB 2347 and when enabling Mesh RF, the AP is locked to 2GHz.

aanon4, 2023-02-20:
Let's keep the AP discussion over here (the other issue is just for the update process).
Pinging you on mattermost to ask some questions.

aanon4, 2023-02-20:
This is what my hAP ac3 looks like with 2347 on it:
[image]

aanon4, 2023-02-20:
If you're up for changing one line by hand to test something, that'd be fantastic?

File: /www/cgi-bin/setup
Currently:
```local alt_phy = phy == "phy0" and "phy1" or "phy0"```
Change to:
```local alt_phy = (not phy or phy == "phy0") and "phy1" or "phy0"```
And see how that changes things

aanon4, 2023-02-20:
https://github.com/aredn/aredn/pull/728

k6ccc, 2023-02-20:
To be honest Tim, editing a 1487 line file in vi is getting beyond my comfort level.  I can barely manage to use vi with an online help on another screen.
Unless the nodes have a more friendly editor.

aanon4, 2023-02-20:
Not to worry - This should take care of the problem and I know you'll test the change once it hits nightly - much appreciated as ever.

k6ccc, 2023-03-05:
Fixed a rev or two ago.

closed 2023-03-05 by k6ccc