aredn/aredn #721: NB 2339 on hAPac3 upgrade file not recognized

k6ccc, 2023-02-18:
Downloaded hAPac3 nightly build 2339 and node advised that the uploaded file is not recognized.  This is the file I used:
aredn-2339-05d247d-ipq40xx-mikrotik-mikrotik_hap-ac3-initramfs-kernel.bin
Deleted the file from my PC and downloaded it again from http://downloads.arednmesh.org/snapshots/targets/ipq40xx/mikrotik/
Same result:
[image]

Next tried the Download Firmware selection.  After the refresh, the selector correctly listed NB 2339.  Selected Download and node advised sysupgrade is not available:
[image]

Note that this node has a WAN connection - not trying to go across the mesh to get internet.

aanon4, 2023-02-19:
Well this is a bit odd. Nothing had changed in the build process think. Lets see if this resolves itself when the next nightly is created.

k6ccc, 2023-02-20:
Same problem with NB 2347
In case it would help, here is a support file...
[supportdata-K6CCC-hAPac3-at-Home-202302200954.gz](https://github.com/aredn/aredn/files/10786906/supportdata-K6CCC-hAPac3-at-Home-202302200954.gz)

aanon4, 2023-02-20:
Question:

You initially said you tried upgrading using ....ac3-initramfs-kernel.bin which isnt the correct file. Was there something else available in the tool? If should be sysupgrade or nand-sysupgrade.

k6ccc, 2023-02-20:
You mean it might work better if I used the right file on the upload?  Well Duh!  Yes, it is upgrading with:
aredn-2347-68f2425-ipq40xx-mikrotik-mikrotik_hap-ac3-squashfs-nand-sysupgrade.bin

However, when trying the Download file option, both a couple days ago and also this morning, I got the:
Firmware CANNOT be updated
sysupgrade is not available
error message as seen in the second graphic above.
I don't normally use the Download option for upgrades...

aanon4, 2023-02-20:
https://github.com/aredn/aredn/pull/727
This should take care of the second part.

k6ccc, 2023-02-20:
Thanks.
Also note that AP locked to 2GHz if Mesh RF still exists on the hAP ac3.
Updated #720

k6ccc, 2023-02-23:
It would appear that this issue is resolved.  I updated to NB 2354 using the upload method.  As expected, that method was required at that time.  However after the 2354 update, the Download method is working to install NB 2361.

Thanks team!

closed 2023-02-23 by aanon4