aredn/aredn #723: supported platform matrix MikroTik model numbers are wrong

cTurtle98, 2023-02-18:
there are mikrotik devices listed with the wrong model numbers on the supported platform matrix
specifically the BaseBox and the mANTBox

`RB912UAG-2HPnD` is just a 2ghz bare routerboard
NOT an `BaseBox`

**`RB912UAG-2HPnD-OUT` is the model number for the `BaseBox 2`**

`RB912UAG-5HPnD` is just a 5ghz bare routerboard
NOT an `BaseBox`

**`RB912UAG-5HPnD-OUT` is the model number for the `BaseBox 5`**

`RB911G-2HPnD` is just a 2ghz DISCONTINUED bare routerboard
NOT an `mANTBox`

**`RB911G-2HPnD-12S` is the model number for the `mANTBox 2 12s`**

`RB911G-5HPnD` is just a 5ghz high power bare routerboard
NOT an `mANTBox`

**`RB921GS-5HPacD-15S` is the model number for the `mANTBox 5 15s`**

**`RB921GS-5HPacD-19S` is the model number for the `mANTBox 5 19s`**

if we do actually support those specific bare routerboards then we need to add a new model section `RouterBoard` and list those models in the appropriate frequency columns

cTurtle98, 2023-02-18:
RB921 is listed correctly on openwrt so I know that isnt an upstream issue
https://openwrt.org/toh/mikrotik/rb921

openwrt only has device pages for the 91x series routerboards themselves and not the products that contain the board.
https://openwrt.org/toh/hwdata/mikrotik/mikrotik_rb911g-5hpnd
https://openwrt.org/toh/hwdata/mikrotik/mikrotik_rb911g-2hpnd
https://openwrt.org/toh/hwdata/mikrotik/mikrotik_rb912uag-5hpnd
https://openwrt.org/toh/hwdata/mikrotik/mikrotik_rb912uag-2hpnd

that could explain those model numbers being listed wrong I suppose if we are just referencing the openwrt build profile

ae6xe, 2023-02-18:
There are many model numbers, with most of the vendors, where the motherboard is the same.  The specific packaging of a device may have further tailoring of the ending numbers of the model string, e.g. ubiquity model numbers end in "US" or "INTL" depending on how the firmware is configured from the factory.  Also, different gain antennas, but identical motherboard and identical firmware. 

What has been done in the past, is chop off the post numbers, such that, for example, any model number with "RB912UAG-5HPnD*" is compatible with the listed firmware.

Yes, upstream openwrt in the latest release is resolving this issue, and will flow through into AREDN's next release (we lag a bit :) ).   

In AREDN's next release the download page will be more precise, or will be replaced with a different and smarter firmware selector feature.  See the AREDN nightly build images -- one image per model already:

http://downloads.arednmesh.org/snapshots/targets/ath79/mikrotik/
http://downloads.arednmesh.org/snapshots/targets/ipq40xx/mikrotik/

cTurtle98, 2023-02-18:
how is this file generated? http://downloads.arednmesh.org/snapshots/SUPPORTED_DEVICES.md

ae6xe, 2023-02-19:
take a look at:
http://downloads.arednmesh.org/snapshots/targets/ath79/mikrotik/profiles.json

Openwrt build generated data, we like to avoid manually typing in to create a firmware selector tool or a download matrix.

aanon4, 2023-02-20:
@cTurtle98 The SUPPROTED_DEVICES.md file is updated by hand. If there are correction and clarifications to this then you could submit a PR for that.

closed 2023-04-01 by cTurtle98