aredn/aredn #726: Allow dual-radio devices to use either for Mesh RF

ab7pa, 2023-02-20:
Allow folks with dual radio devices (hap, ar750, etc) to have a “Band” selector for **Mesh RF** like there is for _LAN Access Point_ and _WAN Wifi Client_.  If the 5.8 ghz radio is not being used for anything else, it could be used as a **Mesh RF** radio.

[image]

aanon4, 2023-02-20:
What happens if people want to use both radios for the Mesh? Or both for the AP? I was talking to people who wanted to do both these things over the weekend.

We do need to make radio use more flexible, but perhaps we need to go further than this?

ab7pa, 2023-02-20:
There are devices on which that would make sense, but currently AREDN doesn't support any of them.  This issue was opened as a possible first step toward a potentially more complex radio usage scenario for future supported dual-radio devices.

ae6xe, 2023-02-20:
I suspect this discussion is about the timing of 1) enabling Mesh RF to choice between the 2GHz or 5GHz band; 2) then, enabling there to be 2 mesh RF interfaces at the same time (instead of 2 devices at a $$$$ tower rental site).    If able to do step 1 before next release would be welcomed in the community.  Step 2 is much more work.

ab7pa, 2023-02-20:
A scenario like this could have immediate utility with currently supported AREDN devices.

[image]

k1ky, 2023-02-21:
See my original issue #37 - this is now a much better possibility. I'll make a note there and consider closing it in favor of this one..

km6lslf, 2023-03-19:
The potential value of this improvement rises greatly with the addition of new devices like the hAP ac3. High gain external antennas on that device paired with the ability to use mesh on both radios would be fantastic

cTurtle98, 2023-03-30:
this can also tie into #7

k1ky, 2023-05-26:
Looks like this is now a reality with Nightly 20230526?  Great job guys!

km6lslf, 2023-05-26:
> Looks like this is now a reality with Nightly 20230526? Great job guys!

For which devices? If I have one that supports this I'd like to try it immediately

km6lslf, 2023-06-23:
When using the 5 GHz radio for mesh, my GL iNET AR750 fails to connect to 2.4 GHz WiFi. Is this occurring for others? 

Running 20230613-52c7286. Support data attached.
[supportdata-KM6SLF-AR750-CAR-36-202306232134.gz](https://github.com/aredn/aredn/files/11853245/supportdata-KM6SLF-AR750-CAR-36-202306232134.gz)

k1ky, 2023-06-24:
> When using the 5 GHz radio for mesh, my GL iNET AR750 fails to connect to 2.4 GHz WiFi. Is this occurring for others?
KM6SLF: Appears that my AR750 (Creta) isn't connecting Wi-Fi as WAN with Nightly 20230623 either. May need to open up a new issue.

km6lslf, 2023-06-24:
I think it may be related to this change, but if the people who worked on
it think it should be it's own issue I can create one. Were you involved in
this feature being added?

aanon4, 2023-06-24:
As this problem is device specific lets open a new ticket. This one will be closed as the feature has been added.

closed 2023-06-24 by aanon4