aredn/aredn #729: GL.iNet AR-750S-EXT not working in nightly build 2347

Rom3oDelta7, 2023-02-20:
I am attempting to image an AR-750S (Slate) using the nightly build aredn-2347-68f2425-ath79-nand-glinet_gl-ar750s-nor-nand-squashfs-sysupgrade.bin.

I loaded the image on the OpenWRT upgrade page and it installs, reboots, and assigns an IP address of 192.168.1.11. At this point, things get weird. First, the ports are not working as I expect. I would normally use the right LAN port at this stage, but it will not come up. Only the middle, DtD, port works. I cannot navigate to the setup page. (Aside: in the 2/4 nightly I could get to setup, but most of the config settings were missing). In the 2347 release, I  immediately get a "Bad Gateway" message. I tried the usual tricks to resolve this, including clearing the DNS cache, clearing the browser cache, a new browser tab, a different browser, rebooting the Ethernet switch, unplugging and reinserting the Ethernet cables, and directly connecting to the device. All with no effect. Of particular interest is the fact that the device type is unknown, which probably explains why it does not get very far.

I tried the other GL-AR750S image (nor instead of nor-nand) and it will not get past the SHA check on the OpenWRT firmware upload page.

Here is the sysinfo screen data:
 
 node: NOCALL-35-0-231
model: GL.iNet GL-AR750S (NOR/NAND)

!!!! UNSUPPORTED DEVICE !!!!
boardid: GL.iNet GL-AR750S (NOR/NAND)
Device HAS BEEN TESTED AS UNSUPPORTED

br-lan 94:83:C4:23:00:E6
eth0   94:83:C4:23:00:E6
eth0.1 94:83:C4:23:00:E6
eth0.2 94:83:C4:23:00:E6
eth0.3 94:83:C4:23:00:E6
tunl0  00-00-00-00-00-00-00-44-00-00-00-00-00-00-00-00
wlan0  94:83:C4:23:00:E7
wlan1  94:83:C4:23:00:E6

/proc/cpuinfo
system type		: Qualcomm Atheros QCA956X ver 1 rev 0
machine			: GL.iNet GL-AR750S (NOR/NAND)
processor		: 0
cpu model		: MIPS 74Kc V5.0
BogoMIPS		: 385.84
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 32
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16 dsp dsp2
Options implemented	: tlb 4kex 4k_cache prefetch mcheck ejtag llsc dc_aliases perf_cntr_intr_bit cdmm contextconfig perf
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

nvram
hsmmmesh.settings=settings
hsmmmesh.settings.wifimac='94:83:c4:23:00:e7'
hsmmmesh.settings.mac2='35.0.231'
hsmmmesh.settings.node='NOCALL-35-0-231'
hsmmmesh.settings.dtdmac='35.0.230'

aanon4, 2023-02-20:
This should add basic support:
https://github.com/aredn/aredn/pull/730
If you can test this change once it's released that'd be fantastic - I dont have access to this device and more work may be required.

aanon4, 2023-03-09:
Is this fixed?

closed 2023-03-28 by aanon4