aredn/aredn #731: CPE710 Channels wrong

wa3lab, 2023-02-20:
I recently purchased and programmed a CPE710 with nightly build 2339.  The node seemed to work OK on channel 180 to connect to my home node.   When I changed it to channel 146 for a remote connection.  It would not connect.  I have since found it is transmitting on channel 149.  I posted this disparity on the AREDN forum and KK4ZUZ suggested I inspect the config in /etc/config/wireless. It showed the programmed channel 146, but the unit transmitted on 149.  As per KK4ZUZ I tried channel 140 and 144 which both worked correctly.  I randomly tried channel 162 which caused the node to transmit on 163.  All these tests were done at 10 MHz bandwidth. I repeated the test on channel 162 at 5 MHz bandwidth and it again failed transmitting on channel 163. If you need any other info please contact me.  Support data file included
73,
[supportdata-WA3LAB-CPE710-202302201905.gz](https://github.com/aredn/aredn/files/10787336/supportdata-WA3LAB-CPE710-202302201905.gz)

Glenn WA3LAB

aanon4, 2023-02-20:
To summaries:

Set -> Actual
140 -> 140
144 -> 144
146 -> 149
162 -> 163
180 -> 180

I'm just trying to understand the failure pattern. I dont suppose you could try channels 145, 147 and 148 could you?

aanon4, 2023-03-09:
Any further information on this problem?

wa3lab, 2023-03-09:
I have been loading the nightly builds since I posted this issue and the problem persists.  I just read Issue #743 and I think it is related.  I tried re-running some tests using 20 MHz BW . At channel 146 in 20 MHz BW, the receiver sees it at channel 146.  If I test it at 10 MHz BW the receiver sees it at channel 149.  Even though it sees the correct channel(in 20 MHz BW), it does not see the correct MAC address nor does it connect. I think Oozil in issue #743 is on to something.  If you have any specific tests you would like run, please let me know.
Glenn

aanon4, 2023-03-09:
I think I misunderstood your original report - is this about a node operating as a hotspot?

wa3lab, 2023-03-09:
No it is not acting as a hotspot, but the fact that it got the channel 
right if the bandwidth is set to 20MHz was somehow related, since I was 
running with a 10 or 5MHz BW. When I run 20MHz BW i see the correct 
channel but the MAC is wrong.

wa3lab, 2023-04-19:
I just installed the latest release firmware in my TP-Link CPE710 and I have rerun the channel test.  The first number is the channel programmed on the setup page, and the second number is where the node is actually transmitting.  All readings were done at 10 MHz  bandwidth. Here are the results.
Set -> Actual
131 -> 132
132 -> 132
133 -> 134
134 -> 134
135 -> 136
136 -> 136
137 -> 138
138 -> 138
139 -> 140
140 -> 140
141 -> 142
142 -> 142
143 -> 144
144 -> 144
145 -> 149
146 -> 149
147 -> 149
148 -> 149
149 -> 149
150 -> 151
151 -> 151
152 -> 153
153 -> 153
154 -> 155
155 -> 155
156 -> 157
157 -> 157
158 -> 159
159 -> 159
160 -> 161
161 -> 161
162 -> 163
163 -> 163
164 -> 165
165 -> 165
166 -> 166
167 -> 167
167 -> 167
167 -> 167
167 -> 167

wa3lab, 2023-04-19:
I just installed the latest release firmware in my TP-Link CPE710 and I have rerun the channel test. The first number is the channel programmed on the setup page, and the second number is where the node is actually transmitting. All readings were done at 10 MHz bandwidth. Here are the results.
Set -> Actual
131 -> 132
132 -> 132
133 -> 134
134 -> 134
135 -> 136
136 -> 136
137 -> 138
138 -> 138
139 -> 140
140 -> 140
141 -> 142
142 -> 142
143 -> 144
144 -> 144
145 -> 149  <-------
146 -> 149
147 -> 149  <-------
148 -> 149
149 -> 149
150 -> 151
151 -> 151
152 -> 153
153 -> 153
154 -> 155
155 -> 155
156 -> 157
157 -> 157
158 -> 159
159 -> 159
160 -> 161
161 -> 161
162 -> 163
163 -> 163
164 -> 165
165 -> 165
166 -> 166
167 -> 167
168 -> 168
169 -> 169
170 -> 170
171 -> 171
172 -> 172
173 -> 173
174 -> 174
175 -> 175
176 -> 176
177 -> 177
178 -> 178
179 -> 179
180 -> 180
181 -> 181
182 -> 182
183 -> 183
184 -> 184

You can see the obvious LSB issue, but channels 145, 146, and 147 don't fit.  Anyway...that was fun. NOT
Glenn WA3LAB

wa3lab, 2023-04-19:
I just installed the new release firmware 3.23.4.0 and re-ran the channel test.  The results are in the Excel file.  Let me know if it helps. 
Glenn WA3LAB  

[CPE710 Channel Matrix.xlsx](https://github.com/aredn/aredn/files/11268205/CPE710.Channel.Matrix.xlsx)

wa3lab, 2023-08-01:
I re-tested The CPE-710 with nightly build 20230722 and this issues still exists.
https://github.com/aredn/aredn/files/11268205/CPE710.Channel.Matrix.xlsx

wa3lab, 2023-08-29:
I retested the CPE-710 with release 23.8.0 and this channel issue persists.