aredn/aredn #733: Mikrotik RBSXTsqG-5-ac not accepting Nightlies.

nc8q, 2023-02-21:
I have tried at least 4 current Nightly Builds attempting to get 
AREDN firmware onto a Mikrotik RB-SXTsqG-5-ac device.
All seems well until after loading/booting the sysupgrade.bin file.
At that point the device does not respond to pings or http: requests on 192.168.1.1
I cannot return to First Boot (http://192.168.1.1) to complete the upgrade.

73, Chuck nc8q

aanon4, 2023-02-21:
I'll have to pull my RB-SXTsqG-5-ac out of the basement to run the full first-install process, but I can upgrade it successfully from a nightly from about a week ago to the current one. This is failing when you're trying a brand new device?

aanon4, 2023-02-21:
Dragged my device out of the basement - wiped it - did the initram install - did the sysupgrade (today's nightly) ... and it's fine. Came back on 192.168.1.1 and everything is installed.

aanon4, 2023-02-21:
And just did the basic config - came back on a 10.X address.

aanon4, 2023-03-07:
I know "it worked for me" isnt useful, so to get past that I'm going to have to ask you to describe, in as much details as possible, all the steps you're taking - starting from powering the device and the state of the device before you being the upgrade process. Thanks (and sorry)

nc8q, 2023-03-08:
gelmce@nc8q-acer:~$ cd /tftp
gelmce@nc8q-acer:/tftp$ ls -l
total 11680
-rw-rw-r-- 1 gelmce gelmce 5770852 Mar  7 08:50 aredn-2397-0f7a9d9-ipq40xx-mikrotik-mikrotik_sxtsq-5-ac-initramfs-kernel.bin
-rw-rw-r-- 1 gelmce gelmce 6161488 Mar  7 08:50 aredn-2397-0f7a9d9-ipq40xx-mikrotik-mikrotik_sxtsq-5-ac-squashfs-sysupgrade.bin
-rw-rw-r-- 1 gelmce gelmce       9 Feb 13 13:13 enp2s0f0
-rw-rw-r-- 1 gelmce gelmce      36 Mar  7 09:05 model.txt
lrwxrwxrwx 1 gelmce gelmce      76 Mar  7 08:51 rb.elf -> aredn-2397-0f7a9d9-ipq40xx-mikrotik-mikrotik_sxtsq-5-ac-initramfs-kernel.bin
-rw-rw-r-- 1 gelmce gelmce     496 Mar  7 10:34 sysupgrade-result.txt
-rw-rw-r-- 1 gelmce gelmce     353 Mar  7 10:32 sysupgrade-result.txt~
-rw-rw-r-- 1 gelmce gelmce     219 Feb 21 20:59 tftp.sh
gelmce@nc8q-acer:/tftp$ cat tftp.sh
ifconfig
sudo ifconfig enp2s0f0 192.168.1.9
ifconfig
sudo dnsmasq -i enp2s0f0  -u gelmce --log-dhcp --bootp-dynamic --dhcp-range=192.168.1.100,192.168.1.200 -d -p0 -K --dhcp-boot=rb.elf --enable-tftp --tftp-root=/tftp/
gelmce@nc8q-acer:/tftp$ sh tftp.sh
enp2s0f0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether dc:0e:a1:0b:b6:0d  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 548  dropped 0  overruns 0  frame 0
        TX packets 89  bytes 15591 (15.5 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 16  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 240  bytes 27259 (27.2 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 240  bytes 27259 (27.2 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlp3s0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 08:11:96:d6:26:e8  txqueuelen 1000  (Ethernet)
        RX packets 181  bytes 19605 (19.6 KB)
        RX errors 0  dropped 3  overruns 0  frame 0
        TX packets 213  bytes 26169 (26.1 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

[sudo] password for gelmce: 
dnsmasq: started, version 2.86 DNS disabled
dnsmasq: compile time options: IPv6 GNU-getopt DBus no-UBus i18n IDN2 DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC loop-detect inotify dumpfile
dnsmasq-dhcp: DHCP, IP range 192.168.1.100 -- 192.168.1.200, lease time 1h
dnsmasq-tftp: TFTP root is /tftp/  
enp2s0f0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 192.168.1.9  netmask 255.255.255.0  broadcast 192.168.1.255
        ether dc:0e:a1:0b:b6:0d  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 548  dropped 0  overruns 0  frame 0
        TX packets 89  bytes 15591 (15.5 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 16  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 241  bytes 27332 (27.3 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 241  bytes 27332 (27.3 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlp3s0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 08:11:96:d6:26:e8  txqueuelen 1000  (Ethernet)
        RX packets 181  bytes 19605 (19.6 KB)
        RX errors 0  dropped 3  overruns 0  frame 0
        TX packets 213  bytes 26169 (26.1 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
[... 98 lines cut ...]
        inet6 fe80::4da1:68aa:8169:226a  prefixlen 64  scopeid 0x20<link>
        ether dc:0e:a1:0b:b6:0d  txqueuelen 1000  (Ethernet)
        RX packets 143  bytes 15809 (15.8 KB)
        RX errors 439  dropped 0  overruns 0  frame 0
        TX packets 288  bytes 40049 (40.0 KB)
        TX errors 4  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 16  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 556  bytes 49175 (49.1 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 556  bytes 49175 (49.1 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlp3s0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 08:11:96:d6:26:e8  txqueuelen 1000  (Ethernet)
        RX packets 24  bytes 3323 (3.3 KB)
        RX errors 0  dropped 1  overruns 0  frame 0
        TX packets 58  bytes 8054 (8.0 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

gelmce@nc8q-acer:~$

aanon4, 2023-03-09:
Thanks for this. A few questions:

1. The device is currently flashed with the Mikrotik firmware, yes? If will boot into that if you're not trying to flash it? I'm trying to establish the initial state of the device.
2. The firefox step is you going to the admin page, selecting the file to download, downloading it, and watching the firmware is updating page until it times out? (you didnt details these steps)
3. Why do you reboot the laptop? How long did you wait before doing that? What there a reason?
4. If you dont reboot the laptop, does the network interface come up? If so, if you do a tcpdump -i INTERFACE_NAME on it, is there any traffic?

Thanks

dman776, 2023-04-19:
has this been resolved?

aanon4, 2023-04-19:
I have never been able to replicate this and I've flash a number of these devices now.

closed 2023-04-19 by dman776

gskuhlman, 2023-04-25:
I have the SXT 5HPnD with the same issue and I've tried about 20 versions but the latest and I can consistently reload the routeros, get it to accept bootp first boot.  Then I've tried going directly to admin page and selecting the sysupgrade file, or done a winscp copy of the file and run the sysupgrad -F -n /tmp/<filename> method and waited and waited but it never comes back after that.   I have computer to dumb switch and switch to SXT and don't switch the computer to DHCP until the sysupgrade is done (again, either from command line or admin page).

I've got about  20-30 hours (not kidding) trying to get this to load but can't get past that initial nocall screen.  I've gotten good at reloading the device's OS or booting from PXE environment.  100% success on getting that step done and booting into AREDN localnode.locl.mesh screen.

One new piece of information today is I tried the command line upgrade and without the -F I got a metadata error.

For clarity I'm loading the file below on the admin page or via command lines sysupgrade
aredn-3.23.4.0-ath79-mikrotik-mikrotik_routerboard-sxt-5hpnd-squashfs-sysupgrade.bin

I've tried going to basic setup screen as well and changing nocall to my call and setting pw.  Same results.  

On the plus side,  it has keep me engaged for WAY more time than the device costs so I've gotten my money's worth of it. ;)

This last attempt tonight shows it stuck in bootp when I look at wireshark.

I would love to troubleshoot the boot process some more if there was some more information on how to do that.
-KF0DYW

more information:
https://www.arednmesh.org/content/mikrotik-sxt-routerboard-boot-settingswont-sysupgrade

dman776, 2023-04-25:
> the sysupgrad -F -n /tmp/ method and waited and waited but it never comes

http://downloads.arednmesh.org/afs/www/?version=3.23.4.0&target=ipq40xx%2Fmikrotik&id=mikrotik_sxtsq-5-ac

aanon4, 2023-04-25:
@gskuhlman Note that this Issue was for the AC version of the SXT not the SXT 5HPnD version.