aredn/aredn #737: hAP ac3: PoE Passthrough Active In “Advanced Configuration” Menu

jksnake, 2023-02-28:
When I graduated from the ac Lite to the ac3, I simply transferred my settings from the Lite, on which I used PoE Passthrough to power the Rocket M5.  It worked fine on the ac3 for about a month but the node has been malfunctioning for a couple of weeks.  It went completely dark this morning so, after turning the feature off, I placed a PoE injector in the Rthernet and everything works fine.

I then remembered hearing that PoE Passthrough is not available on the ac3…but it was!
[image]

Kevin Sherwood, AJ7C

closed 2023-02-28 by jksnake

aanon4, 2023-02-28:
In case anyone comes looking here, the PoE option is available on the AC3 but *not* the AC2. Also worth noting that the amount of power you can get from this PoE out is limited.

jksnake, 2023-02-28:
Thanks for the info, Tim.Kevin Sherwood, AJ7CSent from my iPad