aredn/aredn #747: "Invalid Mesh IP Address" error after Initial Install on MikroTik RBLHGG-5acD-XL

kearneyjf, 2023-03-12:
After installing the Factory file (rb.elf) for nightly build 2414 on a MikroTik LHG XL 5 ac (P/N RBLHGG-5acD-XL), the AREDN header shows up as NOCALL without any extension. I expect to see NOCALL-XXX-XXX-XXX. When the Sysuprade file is uploaded to the node, the node accepts the Sysupgrade file and I am able to connect to the node with a browser using the 192.168.1.1 IP address. If I try to save the configuration with a Node Name and Password, I get an error message saying that I have an "invalid Mesh IP address". The Mesh IP Address displayed on the AREDN screen is 10.0.0.0

Looking at the Support Data file, it shows that the AREDN software is able to read the MAC address of the node. My understanding is that the Mesh IP Address is derived from the MAC address. So, it appears that the AREDN software is not able to translate the MAC address into a 10-XXX-XXX-XXX IP mesh address.

Same problem occurs with nightly builds 2368, 2378, 2390, 2397, and 2405.

A screen shot showing the "Invalid Mesh IP Address" message and the Mesh IP Address of 10.0.0.0 is attached. The Support Date file for the node is also attached.

[MikroTik LHG Screen Shot Showing Invalid IP Address Error 230311.docx](https://github.com/aredn/aredn/files/10949895/MikroTik.LHG.Screen.Shot.Showing.Invalid.IP.Address.Error.230311.docx)

[supportdata-NOCALL-202301030028.gz](https://github.com/aredn/aredn/files/10949896/supportdata-NOCALL-202301030028.gz)

aanon4, 2023-03-12:
We need to add something about this to the install instructions (alerting @ab7pa).

So some nodes take quite some time to initialize on first boot. This related to how the node now need more entropy to set up some things (like generating initial ssh keys). Eventually, if you refresh the page after a couple of minutes, you should see the expected NOCALL-X-X-X and you need to wait for that before proceeding.

kearneyjf, 2023-03-12:
Thanks very much. I’ve installed the software many times on the hAP successfully but no luck on this LHG after many tries.First thing tomorrow I will give that a try and report back,Again, thanks very much for the quick reply.73,John, KM6ZJT

kearneyjf, 2023-03-12:
Hi, Tim.
Tried again. Still didn't work. Here's what I did. Let me know if I'm doing
something wrong.
Set up my computer with fixed IP address of 192.168.1.10
Connected to the node through a dumb switch.
Set up tiinypxe server so that it was ready to load the rb.elf file.
With the ethernet cable unplugged from the node (power off), I pressed
Online on the pxe server.
While holding the reset button on the node, I plugged in the ethernet cable.
After about 15 seconds, I get the "DoReadFile rb.elf" message.
Release the reset button and let file loading progress.
After about another 20 seconds messages begin to scroll down the pxe server
window.
These messages are eventually of a type as shown below:

3:54:28 AM DHCPd:OFFER sent, IP:192.168.1.11, XID:3CA11024

3:54:31 AM DHCPd:DISCOVER received, MAC:DC-2C-6E-DB-40-10, XID:3CA11024

3:54:31 AM DHCPd:OFFER sent, IP:192.168.1.12, XID:3CA11024

3:54:34 AM DHCPd:DISCOVER received, MAC:DC-2C-6E-DB-40-10, XID:3CA11024

3:54:34 AM DHCPd:OFFER sent, IP:192.168.1.13, XID:3CA11024

I was monitoring the node with a browser looking at the 192.168.1.1 IP
address.

Shortly after the "OFFER sent" messages began to scroll down the screen, I
was able to get the "NOCALL" screen to show up in the browser.

I still did not press the Offline button on the pxe server.

I watched this for 15 minutes, refreshing the browser about once each
minute. I never got anything but the original NOCALL screen without any
numerical extension.

After 15 minutes I pressed the Offline button on the pxe server and
uploaded the Sysupgrade file.

Same result as before: Mesh IP address  of 10.0.0.0 and "Invalid Mesh IP
Address" error  message when I try to save the configuration.

Am I missing something? Am I not waiting long enough for the pxe server  to
load the rb.elf file? 15 minutes seems generous.

Are there any clues I should be looking for in the messages on the pxe
server window?

aanon4, 2023-03-12:
Well heck. Thanks for this. I think I have one of these devices in my box of test devices so I'll dig it out and work out what's going on. Will update this with what I find.

kearneyjf, 2023-03-12:
Thanks, Tim.

If you find that don’t have one, I’d be happy to ship you the one I have.

aanon4, 2023-03-13:
It looks like I had support for the RBLHGG-5acD but not the XL model. I'll fix that today.

aanon4, 2023-03-13:
https://github.com/aredn/aredn/pull/750

kearneyjf, 2023-03-14:
Installed Build 2423 on my MikroTik RBLHGG-5acD-XL. Solved the problem.
Thanks very much for the quick fix.
John, KM6ZJT

closed 2023-03-14 by aanon4