aredn/aredn #748: ERR_NAME_NOT_RESOLVED when using GL-inet AR-300Ms

convolved-KY4G, 2023-03-13:
I have two GL-inet AR-300Ms in an AREDN testbed.  The two are connected via the mesh network and can be seen by each other by clicking the mesh status button.  The connection sometimes shows a link of the name of the other node (e.g KY4G-GLAR300 which resolves to http://ky4g-glar300.local.mesh:8080/) and sometimes shows a link to its IP address (e.g. 10.23.85.83 which resolves to http://10.23.85.83:8080/cgi-bin/mesh).  If the IP address is shown, it can be clicked on and successfully goes to the status page of the other node.  If the name of the other node is shown, when you click on the name it says ERR_NAME_NOT_RESOLVED.  This happens consistently between the two nodes.  If you manually enter the IP address directly into a browser, it navigates to the status page just fine.  This seems like a bug.  I am running firmware 3.22.12.0 on both nodes.  There are three images below to further show what I am seeing.

[image]
[image]
[image]

aanon4, 2023-03-13:
Are you running a release or a nightly. This should be fixed in the nightly build.

convolved-KY4G, 2023-03-13:
I was running the latest release.  I thought the nightly build only fixed an issue specific to the AR150s.  I upgraded the nodes and it did indeed fix it for the AR300s too.

closed 2023-03-13 by convolved-KY4G