aredn/aredn #755: Feature request: captive portal

km6lslf, 2023-03-19:
Adding an optional captive portal feature would go a long way in helping mesh nodes at city and county government facilities be used as intended and help avoid frustration with the equipment and local RACES organizations during activation incidents.

It looks like there are some captive portal options for openwrt: https://openwrt.org/docs/guide-user/services/captive-portal/start

OpenNDS looks most promising to me, but my experience in this area is pretty limited.

mathisono, 2023-03-24:
interesting. It would make use of this, as long as: 

-Can be enable and disabled after install 
-Provides a portal for clients connecting to a node in  AP mode.
-option WiFi clients must pass the portal to complete the connection with a node via Wifi.
-option LAN clients must pass the portal to complete the connection with a node to access the Lan.
-The portal displays text on the Portal that can be modified by the control operator.
-The text and hyperlinks fields displays on the Portal, They can be modified by the control operator.

I display a Banner on my imagery page. 

--------
NOTE: "Best Practice Agreement and Terms Of Operation Participating Ham Radio operators by connecting / interacting via RF/high rate data packet to this "mesh" [BAM/SFWEM](https://www.sfwem.net/). An AREDN based network is open to any licensed  Amateur radio operator. By connecting over RF and or Internet tunnel provided by W6BB, as a licensed radio operator you agree to practice and communicate too Best Standards of Operating Practice and Rules identified in Title 47 CFR Part 97. Please don't misrepresent or obfuscate the origin or contents of a data packets. The Mesh network isn't for commercial purposes, nor intended for use in a manner that may be interpreted as detrimental to Amateur Radio or BAM/SFWEM.

NOTE: [All imagery is provided by amateur radio operators participating in BAM / SFWEM](https://www.sfwem.net/)

NOTE: Access to this page is only available on BAM and isn't available outside BAM. 73 de KJ6DZB
--------

km6lslf, 2023-03-25:
All strong feedback, thanks