aredn/aredn #757: UBNT Nanobeam AC Gen2 fail

wu2s, 2023-03-22:
Attempted to load latest nightly builds (2423, 2442) onto UBNT Nanobeam AC Gen2. It is the International version, not the US version.
Followed all the documented steps.
Got an error message: Invalid version 'WA.ar934x. v8. 5.3-42.AREDN-r20028-43d71ad93e

[image]

Orv, 2023-03-22:
Randy, there's a different procedure for factory installs for Ubiquiti 802.11ac devices.  Did you use it?  If not, it's here https://docs.arednmesh.org/en/latest/_downloads/cc242dc167cc98af688b0c1245d47236/Ubiquiti_AC_First_Install_Checklist.pdf
Orv

wu2s, 2023-03-22:
Orv,
Yes, that is the procedure that I used.
I installed the XC 8.7.0 firmware first, then did the rest of the steps in sequence as shown in the screenshot.

-Randy

Orv, 2023-03-22:
I'm confused.  Your hardware is 'XC' version, and you're trying to install some WA firmware update.  I think the file you should be installing is http://downloads.arednmesh.org/snapshots/targets/ath79/generic/aredn-2442-72dd3ae-ath79-generic-ubnt_nanobeam-ac-xc-squashfs-factory.bin

wu2s, 2023-03-22:
OK I willl try that file. 
I was trying to load the firmware labelled *-nanobeam-ac-gen2-squashfs-factory.bin.
Maybe it was built as a WA type firmware only even thought the device I received is running XC factory firmware.

Orv, 2023-03-22:
I'm wrong, you're right.
But the file you loaded seems like it was an original Ubiquiti firmware file.

wu2s, 2023-03-22:
The firmware file *nanobeam-ac-xc-squashfs-factory.bin loads correctly on this device.
Issue closed.

closed 2023-03-22 by wu2s