aredn/aredn #758: Primary IP address not in sysinfo.json interfaces when RF disabled (nightly 2432)

smsearcy, 2023-03-23:
In firmware 3.22.12.0, when a node has RF disabled a wireless interface in the `interfaces` dictionary in `sysinfo.json` will include the IP address, e.g.:
[image]

However, in nightly 2432, none of the `wlan#` interfaces include the IP address, despite it showing as the "primary address" on the Mesh Status page (and that is the IP address that I am connecting to it via).

This is causing issues for my Mesh Info polling script when it is collecting the information about those nodes.  Not sure if this was an oversight or intentional.

aanon4, 2023-03-23:
It's a bit odd that there was an IP address attached to a wifi interface when the wifi was disabled. In release and nightly, both add the primary IP to a different dummy device when wifi is disabled (but it is done differently). Is this on an hAP Lite? I'd like to reproduce this.

smsearcy, 2023-03-24:
It looks like its happening on about eight nodes in the broader Oregon mesh (and some older development versions than I first noticed).

* MikroTik RouterBOARD 952Ui-5ac2nD (hAP ac lite) - nightly 2405-692fc8e
* MikroTik RouterBOARD RBD52G-5HacD2HnD - nightly 2432-51878e2
* MikroTik RouterBOARD RBD52G-5HacD2HnD - nightly 2316-214a933
* MikroTik RouterBOARD 952Ui-5ac2nD (hAP ac lite) - nightly 2368-d1dba1f
* MikroTik RouterBOARD 952Ui-5ac2nD (hAP ac lite) - nightly 2414-21115f7
* Ubiquiti NanoStation M2 XW - nightly 2405-692fc8e
* MikroTik RouterBOARD RBD52G-5HacD2HnD - nightly 2414-21115f7
* MikroTik RouterBOARD SXT 5nD (SXT Lite5) - nightly 2414-21115f7

None of those have the "primary IP address" associated with a interface in `sysinfo.json`.  All the nodes I'm polling that are running release 3.22.12.0 (and prior) have the IP address available.

Thank you for looking into this.

aanon4, 2023-03-24:
Thanks.
https://github.com/aredn/aredn/pull/760

smsearcy, 2023-03-30:
Unfortunately, none of them are my nodes.  I'll see about reaching out to the owners to update to the latest nightly.  Which version should have the fix?

smsearcy, 2023-04-01:
@aanon4, a MikroTik hAP ac2 was upgraded to 2489-10d99b2, looks like it is now showing up as `br-nomesh`.  Is that what you expected?

[image]

aanon4, 2023-04-01:
Yes.

dman776, 2023-04-19:
can this be closed?

smsearcy, 2023-04-19:
Yes, the issue has been resolved.

closed 2023-04-19 by aanon4