aredn/aredn #763: Gateway error when going to SETUP

KenHorse, 2023-03-25:
I just flashed 2 PBE-5AC-400 with the latest nightly build (dated today). All went fine but when trying to get to SETUP, I'm prompted for credentials (which I enter) and then I'm greeted with a "Bad Gateway The process did not produce any response" error.

This is happening on 2 of 2 units

aanon4, 2023-03-25:
Can you post a screenshot of the main status page immediately before hitting 'setup'?

aanon4, 2023-03-25:
Note that this hardware is not in the support list.

KenHorse, 2023-03-25:
[image]
[image]

KenHorse, 2023-03-25:
Oh? There's a difference between it and a Gen2?

aanon4, 2023-03-25:
Please click on the text reading "!!!! UNSUPPORTED DEVICE !!!!" and post the information it should display. It's probably easy to add this.

KenHorse, 2023-03-25:
Oh duh..the banner..

node: NOCALL-80-102-111
model: 0xe7f5

!!!! UNSUPPORTED DEVICE !!!!
boardid: 0xe7f5
Device HAS BEEN TESTED AS UNSUPPORTED

br-lan F0:9F:C2:51:66:6F
eth0   F0:9F:C2:51:66:6F
eth0.1 F0:9F:C2:51:66:6F
eth0.2 F0:9F:C2:51:66:6F
tunl0  00-00-00-00-00-00-00-44-00-00-00-00-00-00-00-00
wlan0  F0:9F:C2:50:66:6F

/proc/cpuinfo
system type		: Atheros AR9342 rev 2
machine			: Ubiquiti PowerBeam 5AC Gen2
processor		: 0
cpu model		: MIPS 74Kc V4.12
BogoMIPS		: 266.64
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 32
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16 dsp dsp2
Options implemented	: tlb 4kex 4k_cache prefetch mcheck ejtag llsc dc_aliases perf_cntr_intr_bit perf
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

nvram
hsmmmesh.settings=settings
hsmmmesh.settings.wifimac='f0:9f:c2:50:66:6f'
hsmmmesh.settings.mac2='80.102.111'
hsmmmesh.settings.node='NOCALL-80-102-111'
hsmmmesh.settings.dtdmac='81.102.111'

KenHorse, 2023-03-25:
BTW, the firmware is clearly shown at the selector webpage. That's where I got it

aanon4, 2023-03-25:
While I see "Ubiquiti PowerBeam 5AC Gen2" and the "Ubiquiti PowerBeam 5AC 500" in the selector, I do not see the 400 variant listed.

KenHorse, 2023-03-25:
Am I mistaken that the hardware reports itself as "machine : Ubiquiti PowerBeam 5AC Gen2"

aanon4, 2023-03-25:
That's reporting what the firmware loaded onto the device is, not the device itself which is identified by the hex code boardid.

KenHorse, 2023-03-25:
I saw /proc/cpuinfo and assumed it read the info from there

OK, well guess I'll wait for the built

aanon4, 2023-03-25:
https://github.com/aredn/aredn/pull/765

aanon4, 2023-03-28:
A build should now be available. If you could verify that'd be fantastic.

KenArck, 2023-04-14:
Really sorry for the delay. Yes, all is working fine with the new build

closed 2023-04-14 by aanon4