aredn/aredn #768: UBNT Nanobeam AC Gen2 - Unspported Device

wu2s, 2023-03-29:
After closing issue #757 I browsed to the device to begin configuring it.
Device showed Unsupported device message this is detailed below. Attempts to get to the Setup page resulted in error message "Bad gateway"
[image]

Orv, 2023-03-29:
Randy, the screenshot shows it as a Nanostation AC Gen1.  That may be different enough from the Gen2 to cause these symptoms,

aanon4, 2023-03-29:
I'm building a new image for this device as we speak.

aanon4, 2023-03-29:
New build submitted here - https://github.com/aredn/aredn/pull/770
Would be good to get this tested as its a bit of a guess in places.

wu2s, 2023-03-29:
So what it shows INTERNALLY is not what it shows externally.
The box it came in and the device itself are labelled as Nanobeam AC GEN2

Weird. Look at that test date: April 25, 2017 

-Randy

￼

￼

￼

aanon4, 2023-03-29:
The name we display is what device firmware we load onto the device, not what the device actually is. Bit confusing.

wu2s, 2023-03-30:
Tried loading the new (NB 2475) Gen 2 build, That did not work. Loaded the NanoBeam AC Gen 1 firmware and that was successful. Able to configure and run normally. Very annoying that the labels on the device show Gen2 very clearly but this is actually a Gen 1 device.

closed 2023-03-30 by wu2s

aanon4, 2023-03-30:
Hmm. I should still make that gen 2 build work. How did it fail?