aredn/aredn #773: GL.iNet AR-750s (slate) Ethernet ports are mapped incorrectly (nightly 2468)

Rom3oDelta7, 2023-03-30:
I was successful in installing and setting up a GL.iNet AR-750s with nightly build 2468. Thanks for the fix that enabled this device!

Early on in the testing, and everything works so far, including the tunnel client, with one exception: the Ethernet ports are mapped incorrectly. In the GL-AR750 (Creta) box, the ports left to right are: WAN, DtD, LAN. However, the GL-AR750s ports L-to-R are: WAN, **LAN, DtD** (2 & 3 are reversed). I tested connecting 2 devices over DtD and confirmed the rightmost port is indeed DtD. Obviously, as the AR-750 Creta port mapping is documented, the AR-750s (Slate) should use the same order. 

Thanks.

[supportdata-KG7LMI-OFFICE-2-202303291930.gz](https://github.com/aredn/aredn/files/11107173/supportdata-KG7LMI-OFFICE-2-202303291930.gz)

Rom3oDelta7, 2023-06-12:
This problem still exists in 3.23.4.0.

aanon4, 2023-06-12:
Which specific build image are you installing on your AR-750s (Slate)?