aredn/aredn #775: Small sysinfo.json change

r1de, 2023-04-01:
I've been working on some new scripts to poll the network for my map and ran into a weird issue.
I'm pulling sysinfo.json's `link_info` section from each node (among other things) and dumping the info into SQL.
The json array looks like this:
[image]
See the `signal` key at the end?
Turns out `SIGNAL` is a "reserved word" in SQL!
Could we change it to just `sig`, `rf_sig`, _something_ other than the word signal?
Yes, I could change it programmatically, but others may run into this same thing in the future, and it may as well get changed now.

dman776, 2023-04-01:
Can you Alia’s your sql table and prefix it?

Ie.  Select a.signal from my table as a

r1de, 2023-04-01:
Could do that, but it's not on the SELECT, it is on the INSERT.
I think there might be a way to alias a column with another name so it will SELECT/INSERT as either, I remember _something_ like that from my old days of being "kind of the DBA", but I can't find how to do it yet.

r1de, 2023-04-01:
I think I remember we were using MSSQL way back in my old admin days, that may be why I remember we could do that weird stuff that just does not fly in MariaDB/MySQL. :man_shrugging: 

If I could push to github I would change it myself (my account is having troubles it seems, can't push to my repos)!!

smsearcy, 2023-04-01:
You can use a keyword as a column name in MySQL by putting it in double quotes according to [SO](https://stackoverflow.com/a/2889884/3641818).

I'm already using this "signal" attribute from JSON in my parser, would prefer not to see it change.

closed 2023-04-01 by r1de