aredn/aredn #778: RBLHGG-5acD-XL not able to open other nodes

DaleJag, 2023-04-06:
I have installed: aredn-2058-5cff46a-ipq40xx-mikrotik-mikrotik_lhgg-5acd-initramfs-kernel & Upgrade on this node.  It will not take:  aredn-2475-4761380-ipq40xx-mikrotik-mikrotik_lhgg-5acd-xl-initramfs-kernel.  I even tried it as a first install.  The page would go to the rebooting process, but the node lights would never flash.  Only the power and USB lights come on.  No others.  With the mikrotik_lhgg-5acd file loaded, the dish can see the 500+ nodes and list them in the mesh status, however if you click on any of the nodes, it will not bring up that node.  Also, we are sharing the internet on the mesh and this node will not allow it to be seen on my computer.  
[image]
[image]
[image]
[supportdata-WA6MZW-LGH5AC-202301030107.gz](https://github.com/aredn/aredn/files/11174799/supportdata-WA6MZW-LGH5AC-202301030107.gz)

aanon4, 2023-04-06:
Before looking at anything else we need to get your on the current release. It doesnt necessarily surprise me that upgrades between nightly might not work in all cases as we've changed the names of LLHG devices during this process. 

I'm not sure why you cannot install the software using the RAM then SYSUPGRADE process. Can you retry? I did this just the other day without a problem.

DaleJag, 2023-04-07:
This is the first time that the firmware loaded properly.  Never before did it reboot by itself.  
It now will see other nodes and download it's page.  Internet works on it.  
However, only the power and USB lights are on.  No other lights are flashing, 
The software 5acD-XL still will not load.  I tried 5 times to load the kernel file using the pxesrv program.

[supportdata-WA6MZW-LGH5AC-202304062353.gz](https://github.com/aredn/aredn/files/11175095/supportdata-WA6MZW-LGH5AC-202304062353.gz)

aanon4, 2023-04-07:
I'm not sure I understand about the 5acD-XL software. It will not load when you're running the non-XL software already. If you want to switch to that you need to set the "Dangerous Firmware" option in the advance configuration page before doing the upgrade. Does that not work?

I dont think anyone has attempted to make the lights work on this device. If you want to file an Issue specifically for the lights please do.

DaleJag, 2023-04-07:
The  RAM + SYSUPGRADE using the XL version did not work.  However, doing the "Dangerous Firmware" way did work.

Now with the correct firmware in there, the lights still do not flash.  Only power and USB lights are firmly on.  I will file an issue on that.