aredn/aredn #780: MikroTik SXTsq Lite2 support beyond 3.22.12.0

AJ6GZ, 2023-04-07:
SXTsq Lite2 (RBSXTsq2nD) not yet supported post-3.22.12.

AJ6GZ, 2023-04-11:
Upgrade from 3.22.12.0 to 2517 via UI fails due to node reporting itself as 'rb-lhg-5nd' under 3.22.12 (Confirmed on two nodes).
Was able to upgrade via sysupgrade -F.  Once upgraded it shows as 'routerboard-sxt-2nd'.

Other than that it appears stable in a live environment.

----- BEFORE
(Status page)
model:  MikroTik RouterBOARD RBSXTsq2nD

(Administration page)
Current Version: 3.22.12.0   Hardware Type: (ar71xx/mikrotik) mikrotik (rb-lhg-5nd)

root@AJ6GZ-Portable1-SXT2:~# sysupgrade -n /tmp/rb.bin
Device rb-lhg-5nd not supported by this image
Supported devices: mikrotik,routerboard-sxt-2nd rb-sxt-2nd
Image check failed.

----- AFTER
root@AJ6GZ-Portable1-SXT2:~# sysupgrade -F /tmp/rb.bin
Device rb-lhg-5nd not supported by this image
Supported devices: mikrotik,routerboard-sxt-2nd rb-sxt-2nd
Image check failed but --force given - will update anyway!
Saving config files...
Commencing upgrade. Closing all shell sessions.

(Status page)
model:  MikroTik RouterBOARD SXT 2nD (SXT Lite2)

(Administration page)
Current Version: 2517-e5a0b43   Hardware Type: (ath79/mikrotik) mikrotik (routerboard-sxt-2nd)

Orv, 2023-04-14:
This image now installs and behaves as expected.  This issued may be closed.

closed 2023-04-14 by aanon4