aredn/aredn #786: Feature Request: Provide activity indicator for Non-DHCP assigned nodes

k1ky, 2023-04-15:
Request for status indicator  and inclusion to the connected devices list for Fixed IP devices that are connected to a node, similar to the DHCP connected list at the bottom of the Port Forwarding, DHCP and Services GUI Page.  Possibly also provide activity indicator "dots"  similar to the Advertised services indicator dots for services that are not active.  Suggest "Blue" dot for the active devices, and whatever color is being used for inactive services. in both the DHCP Address Reservations and Advertised Services sections of the screen.  As of now, I don't believe there is any indicator for Fixed IP devices to be recognized as actively connected to a node.

dman776, 2023-04-19:
need more detail.  use case maybe?

k1ky, 2023-04-19:
How can we determine what Fixed IP (non-dhcp) nodes are connected that have a reservation? Without a computer connected to the node that has network analysis software, I don't see how you can see what is connected.  Would like to have a way for a node to indicate when something (a fixed IP device) is connected to it.  DHCP connected nodes show up in a list below the DHCP Reservation section.

[image]

k1ky, 2023-04-19:
Use case is that I have my PBX server connected as a Fixed IP device and I have reserved that address, but there is no indication available to show that it is connected.

Orv, 2023-04-20:
Why use a fixed IP address, Tom?  We use a DHCP address for our PBX, and even if it changes, all the phones know the PBX by its hostname, so no biggie.

k1ky, 2023-04-20:
Orv - that's one of those things that's the way it has been.  I have been thinking and there is probably no reason we couldn't switch it to DHCP. I also have a Packet APRS node that we may have set up that way as well, but again, should be able to set it to DHCP as well. I'll discuss this with N2MH regarding his thoughts on the PBX.

aanon4, 2023-04-20:
You can just create a DHCP reservation for this, even if you're not actually using it. Just select the IP address and insert the mac address for the device.

k1ky, 2023-04-21:
That is the procedure that we are actively using for fixed IP devices.  Example the Belmont-PBX device listed above is a fixed IP static server.  Just looking for a way for the node to show the active connection.

ae6xe, 2023-04-22:
A fixed IP address can occur for some older cameras.   What happens is if the camera is rebooted, or anytime the camera is renewing the DHCP lease, and the node is offline,  the camera reverts to a default IP address, e.g. could be 192.168.x.x. or even the reserved mesh node IP address.  Bottom line, it doesn't renew the DHCP lease and stays with a static address.     @aanon4, I think this was handed with the service detection logic, so the camera advertisement is not lost.  With other LAN device status information, handle as time permits?

aanon4, 2023-04-22:
To return to the activity indicator part of this discussion, what would that be? In the PBX example, it isn't connected to the node in any way. Its address is part of a subnet which the node will attempt to route traffic to if its received it - but there's no connection as such. IP is connectionless after all. We could do something with the ARP table, but that's only going to show things which have activity - if no ones talked to a device for an hour then it's not going to show up .. even if it's there. I think it was suggested that we just ping all addresses in the subnet to see if anyone replies. It's an option.

k1ky, 2023-04-22:
Ah but the PBX is connected to the node (UBNT ARHP) Network port 1 and has been set to 10.84.186.50. It has a good number of IP phones registered to it that are active throughout our MESH network.  So if there were a ping interrogation from time to time, it would answer.  How does the DHCP connected list work? Would it be possible to include/test anything that was in the DHCP reservation list that is static?