aredn/aredn #789: Mikrotik list of products shown in AFS omit SXT boards

Orv, 2023-04-17:
When using AFS 1.0 to locate appropriate software, the list of Mikrotik products doesn't list any SXT devices.   They're in AFS; you can find them by just typing sxt.  But they're somehow not linked to the Mikrotik brand:

[image]

aanon4, 2023-04-17:
As the AFS will only show so many matches, it's better if you're more specific in what you type:

[image]

aanon4, 2023-04-17:
Visually, the ... at the end of the list it trying to indicate there's more which isn't shown.

Orv, 2023-04-17:
Explained this to the user.  Closing ticket.

closed 2023-04-17 by Orv