aredn/aredn #790: TP-Link CPE210 v3 listed as unsupported

gerner, 2023-04-17:
I just got one from Amazon. Box says version 3.2. I tried to flash both the latest firmware (aredn-3.23.4.0-ath79-generic-tplink_cpe210-v3-squashfs-factory.bin). Got this error.

sysinfo is as follows:

```
 node: NOCALL-203-61-200
model: TP-Link CPE210 v3

!!!! UNSUPPORTED DEVICE !!!!
boardid: TP-Link CPE210 v3
Device HAS BEEN TESTED AS UNSUPPORTED

br-lan 9C:53:22:CB:3D:C8
eth0   9C:53:22:CB:3D:C8
eth0.1 9C:53:22:CB:3D:C8
eth0.2 9C:53:22:CB:3D:C8
tunl0  00-00-00-00-00-00-00-44-00-00-00-00-00-00-00-00
wlan0  9C:53:22:CB:3D:C8

/proc/cpuinfo
system type		: Qualcomm Atheros QCA9533 ver 2 rev 0
machine			: TP-Link CPE210 v3
processor		: 0
cpu model		: MIPS 24Kc V7.4
BogoMIPS		: 432.53
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 16
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16
Options implemented	: tlb 4kex 4k_cache prefetch mcheck ejtag llsc dc_aliases perf_cntr_intr_bit perf
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

nvram
hsmmmesh.settings=settings
hsmmmesh.settings.wifimac='9c:53:22:cb:3d:c8'
hsmmmesh.settings.mac2='203.61.200'
hsmmmesh.settings.node='NOCALL-203-61-200'
hsmmmesh.settings.dtdmac='203.61.200'
```

gerner, 2023-04-17:
I couldn't get the setup page to load, it went to bad gateway after putting in default username/password and then it wouldn't load any of the pages. I had successfully flashed a different (gl-inet) device earlier in the day, so maybe this is an unrelated browser cache issue?

gerner, 2023-04-17:
I notice that in https://github.com/aredn/aredn/blob/main/files/etc/radios.json#L95 there's an entry for "TP-LINK CPE210 v3" but none for "TP-Link CPE210 v3" (note capitalization). I bring it up because there are different capitalization entries for v2. Is there just a missing capitalization issue?

aanon4, 2023-04-17:
Was just writing this - yes, that's the problem.

aanon4, 2023-04-17:
If you wanted to change this file on your node by hand, you can probably proceed once that's done (I'll make a change in the sources)

gerner, 2023-04-17:
fwiw, I just downloaded aredn-3.22.8.0-ath79-generic-tplink_cpe210-v3-squashfs-factory.bin and flashed it to my unit and everything worked, including getting rid of that bad gateway. Is the bad gateway I was getting something trivially related to the missing entry in radios.json? otherwise perhaps there's something else up with this hardware on AREDN v3.23?

aanon4, 2023-04-17:
The bad gateway is just because the device wasn't recognized by the software - nothing else.

gerner, 2023-04-17:
cool. thanks for the quick response and info!

closed 2023-04-19 by dman776