aredn/aredn #802: Tunnels all down after upgrade to 3.24.3.0

aa3jc, 2023-04-21:
Device:  MikroTik RouterBOARD 952Ui-5ac2nD (hAP ac lite)
All configured Tunnel Server/Clients have now gone dark after firmware upgrade 3.24.3.0    This happened also with the previous firmware upgrade, and was able to recover by retyping in the server side info, and having client disable re-enter data, enable, save and tunnel would re-activate.  Now with 3.24.3.0 nothing is bring back the tunnel clients........  Any Suggestions?!
[AA3JC-HAPAC-HATFIELD-PA-QTH-01.local.mesh8080.pdf](https://github.com/aredn/aredn/files/11299434/AA3JC-HAPAC-HATFIELD-PA-QTH-01.local.mesh8080.pdf)

aanon4, 2023-04-21:
Can you provide the supportdata for the tunnel server in question?

VA2XJM, 2023-04-21:
I had the same problem with 2 or 3 tunnels. While still able to log into the router in front of them, I was able to disable and re-enable the network interface and everything came back alive after that. One location where I do not have access, I needed to ask someone to pull our the WAN cable from hAP and plug it back. It is like the network is not raising fast enough at boot.

aanon4, 2023-04-21:
The "must disconnect the WAN for the hAP to come up" bug is fixed with 3.23.4.0 (once you actually have it running) .. finally! This had been a problem for a long time but only recently has the necessary api been implemented in Linux to allow us to fix it.

VA2XJM, 2023-04-21:
> The "must disconnect the WAN for the hAP to come up" bug is fixed with 3.23.4.0 (once you actually have it running) .. finally! This had been a problem for a long time but only recently has the necessary api been implemented in Linux to allow us to fix it.

I never experienced this bug, my hAP always booted with WAN connected without problems since I use AREDN.

For this upgrade, the hAP booted perfectly and was operating on DtD side, so for the ones with RF link, I was able to access the GUI and find out there was no WAN connection detected. So I logged into the routers in front of the nodes to toggle off and then on the network interface where the WAN is connected and the hAP acquired IP from DHCP and everything was back. Out of 10 tunnel I only had 3 requiring this.

aa3jc, 2023-04-21:
Also noticed that the firmware update changed the Tunnel Server to Client IP address from what it was prior to the upgrade.
[Tunnel_Server_supportdata-AA3JC-HAPAC-HATFIELD-PA-OMNI-01-202304211839.gz](https://github.com/aredn/aredn/files/11299698/Tunnel_Server_supportdata-AA3JC-HAPAC-HATFIELD-PA-OMNI-01-202304211839.gz)
[Client-supportdata-AA3JC-HAPAC-HATFIELD-PA-QTH-01-202304211827.gz](https://github.com/aredn/aredn/files/11299699/Client-supportdata-AA3JC-HAPAC-HATFIELD-PA-QTH-01-202304211827.gz)

k1ky, 2023-04-23:
One thing of note is the length of your node name.  Should be 22 characters or less for reliable Tunnel Client connection to other servers.  Sounds like your IP addy may have ended up being the same as another node on your network?  I had this happen long ago and caused apparent node name duplication.

aa3jc, 2023-04-24:
So Far, No Joy on any of these suggestions, but thanks for the assistance!  This Tunnel issues is affecting 80 Nodes, 3 Pennsylvania county's and 2 phone pbx's that are co-located on the mesh along with many other services for emergency services. We are desperate to get the tunnels working again to resume services to the community's we serve.  
I've in the wait time for Aredn to work on this have re-flashed 3 nodes to older firmware with the hopes this would get us the tunnels back but to no avail. Also changed the host names to much shorter names for testing purposes, but again, no joy.  Looks like we will have to wait for Aredn to address this problem.

mpkleine, 2023-04-24:
Greetings, I have solved my issue.
My WAN IP address changed on the AREDN tunnel server machine from .54 to .56. 
A static DHCP server sets this address on the edge router.
I changed the port 5525 tunnel port forward to this new .56 address, and the system came to life.
My guess is that the WAN MAC must have changed during the update, and the DHCP system assigned a new address to the new MAC.

aanon4, 2023-04-24:
I'm afraid some MAC address might have changed on some devices during this upgrade, so if you had configuration tied to a previous MAC address it would need updating. Sorry for not calling this out. LAN, WAN and DTD now have unique MAC addresses (one some devices the address had been shared) which could cause problems on some switches and routers.

mpkleine, 2023-04-24:
Tim, that would match the behavior that I saw. Since the AR150 shares the three networks on a pair of ports, your change in operation would have hit my DHCP router with a new MAC. My DHCP server then provided a new IP address for this new "device", and the router happily dumped inbound packets to the "old" IP address. Easy to fix, but tough to locate. You're right that documenting this could have sped up recovery. Thanks for the effort!

aa3jc, 2023-04-26:
I have performed all the recommendations above without any success.  I have even taken two nodes (both Mikrotik HAP AC Lites) off-line, taken them into my lab, reflashed them both with the factory build via TFTP to version 3.21.4.0  then preformed an upgrade to version 3.21.4.0 sysupgrade just like you would for a brand new node.  

On RF they work perfectly but "No Joy" getting the Tunnels to work again. And just for no reason, I put both units back on my mesh network.  My domain is pointed back to my home network and works perfectly as I can use my cell phone using mobile data (wifi turned off)    and I have no problem opening my home page that lives on my Synology NAS here at home.....so....

Next, I upgrade both nodes to version 3.23.4.0 rechecked  that the upgrade did not change any IP address for basic settings along with the tunnel server page,  and still no joy with the tunnels.

After that, I took two brand new nodes flashed to 3.23.4.0 and setup a 2nd ISP using a dynamic dns forwarding domain to it, and used my second internet ISP connection that is seperate from my main verizon ISP, and again NO JOY getting the tunnels to function again.

We are only a year or two into getting Mesh going here in the local 3 county's and due to the distance between people with working nodes, RF was not completely viable, hence having to use Tunnels. We have/had 80 nodes of which 20-25 are connected via RF and the rest were tunnels but we have intrest from new hams intrested in putting up a few nodes that would have filled in between and would have eliminated some of the tunnels with RF.   

Last week I responded to the email asking for Support Data with a copy of the data from the node running the Tunnel Server along with one of the  Tunnel Client Support Data, and not sure if it was even received as there has been no receive reply or comments from anyone.

VA2XJM, 2023-05-04:
I have one node I updated to nightly 2637-58d662f on Monday that is in a strange problematic state regarding tunnels...

Tunnel clients (I have two) will not connect to remote server (usual setup for this location). We see into the logs that it tries to connect but nothing on the remote side.

Tunnel server will not allows me to set DNS name (Once saved, comes back empty with invalid dns name error message) and the Network IP will change at every reboot of the device. I've set a tunnel server for testing. After every reboot, I must set back the old network IP, When done, I will see the test tunnel is connected (incoming connection).

This node is remote, so I do not have possibility to do a fresh start soon. I may try latest nightly...

aa3jc, 2023-05-08:
Check your nodes name.  We have come to find out, that if you set the name as per Arden instructions, the node name will support up to 64 characters.......BUT if you are going to activate Tunnels as per Aredn the node name needs to be 22 characters or less.  We have found out this also is not exactly true.  After many bench tests, any name over 20 characters may not work.  At 20 characters, you have a 50/50 chance of it connecting but may or will fail if any changes are made including the channel or bandwidth.  When the nodes comes back up, no tunnels and no matter what you do to fix it, will not work.
If node name is UNDER 20 characters the server / client will like it and perform as it should.  With node name at 18 characters, both the tunnel / client will like it and will reconnect even if you make changes to either node (with an exception to a name change that will require an update to server in the case of a client name change).  
So far, I am not able to "break" the tunnels with v.3.23.4.0

aanon4, 2023-05-08:
@aa3jc I'm still trying to work out what's going on with your hAPs. I wish I had better news for you, but I'm afraid all i can do is ask for more information. Specifically ... with your two hAPs enabled and on your network, can you verify that the main status page on each node shows a WAN address and default gateway that you'd expect (lower/left of the page)? Can you verify that these addresses dont change on reboot? Finally, can you verify that, whichever one you have designated as the tunnel server, has traffic correctly forward to it from your firewall/router?  One way to do this is to telnet to the tunnel port using the tunnel servers internet name. For example, if I was testing this on my own server I'd type the following:
```
telnet aredn.xojs.org 5525
```
This would connect and send a snippet of text containing the word 'vtun'.

aa3jc, 2023-05-21:
AANON4,
Both Haps have static assigned IP's via my router, so the wan address does not change, even on reboot. Tunnel Server is "AA3JC-HAPAC-QTH-01" and Tunnel Client is "AA3JC-HAP-LANSD-1".  All port forwarding from my routers has not changed in 2 years, has work on previous firmware versions, and even 3.23.4.0 but anytime any additions or changes on tunnel server or client takes the tunnels down.  We have seen this behavior on many other server/client nodes, not just this one.  Also, have seen the following issues”

1.  Primary RF address gets changed just for rebooting the node, or doing a version upgrade on the last three firmware upgrades and/or since you changed over to Lua....

2.  It makes no difference if you click “Save Settings” when doing an upgrade.  After it reboots, your RF address can be different than it was originally, along with the Tunnel Server IP Scheme can and will be different, blowing up all the clients and requires sending out email with new settings.  Sometimes it required changing the Tunnel IP 2 or 3 times to get it working again.
b.  I since have added a test node that I can use to test if the tunnels are going to work with the Tunnel Server IP before I send out emails.

3.  I have seen the tunnel server IP scheme mysteriously change during a version upgrade again since Aredn changed over to Lua.  Again, I have seen this happen on other nodes, and even can happen on a simple reboot. 
 
4.  We are seeing that on the Tunnel Server and Tunnel Client pages, some weird things happing like when you click on “Refresh” after doing a “Save” and getting the confirmation at the top, that the page does NOT refresh, and repeated clicks on it has no effect, and requires you to click in the browser address bar and refresh it there.
I debated as to whether I should open a bunch of new tickets, but will not as it all is related to this original issue. 
Joe-AA3JC
[image]
[image]

VA2XJM, 2023-05-22:
Just flashed a brand new hAP AC2 and tunnel will not work. Second hAP AC2 to be flashed later this week.

Cannot set DNS name and tunnel IP network will change every time the page is reloaded. Would support file be handy?

aanon4, 2023-11-15:
Please reopen if this is still happening. I've had no reports of it.

closed 2023-11-15 by aanon4