aredn/aredn #813: Setup of hap AC3 with 3.23.4.0 problems

SensorsIot, 2023-05-01:
I flashed my new hap ac3 first with several nightly builds, and now with the official release. I always encounter the same problem:
In setup, I get this error while saving:

nv parameter 'MAC2' in file '/etc/config.mesh/_setup.default' does not exist

I saw a difference to my old hap device: The menu "LAN Access Point" is not available with the ac3 model
Also, the IP address in "Mesh RF (2GHz)" is left without an address derived from the MAC address and the name is not automatically populated.

[image]

I tried to install the upgrade file with the GUI and with SSH.

73 de Andreas, HB9BLA

SensorsIot, 2023-05-02:
I agree with your judgment that something went wrong with my first nightly build, However, I do not remember which one it was.
Unfortunately, I many times started from scratch. I also checked if I was logged into a "basic" system after I flashed the rb.elf file (the /etc directory was more or less empty, for example). The version of the rb.elf file is also displayed in the terminal when I changed between the different versions.
However, I never was able to check if the update really was successful or if it was aborted and left the old files (or parts of them) in the memory.
Is there a possibility of deleting all files before I start from scratch (or the ones related to ARDN)? And is there a log of the upgrade process available after upgrading? Or how can I otherwise check if the upgrade was successful?

aanon4, 2023-05-04:
You should start again with this node, just as if you'd taken it out of the box and never put AREDN on it. Please follow these instructions again - https://docs.arednmesh.org/en/3.23.4.0/arednGettingStarted/installing_firmware.html#mikrotik-first-install-process (skipping the initial Protected Routerboot part).

tjardick, 2023-05-05:
I took a new HAP AC3 out of the box and had the same issue with current release 3.23.4.0.

I tried performing the same install with the nightly build (currently 2646) but this give a Bad gateway error on the status page and on the basic setup all the WiFi details are not even showing anymore on this page.

I have ahAP ac Lite which i flashed with 3.23.4.0 and it worked out of box. 
I also tried a hAP ac2 but this one takes the rb.elf but after the upload of the sysupgrade.bin the system stays stuck in a boot loop. So maybe it is related to the  ipq40xx devices...

SensorsIot, 2023-05-05:
I ordered a second ac3 to make sure that I start from scratch. One of my friends was able to flash the ac3. But he started with a different nightly build. I do not know if any files in the file system survive the rb.elf process and if we can be sure we really start with a "clean slate". Maybe somebody can confirm?

tjardick, 2023-05-05:
Andreas, does your friend still has the build files? I could test it on my ac3 if you like.

aanon4, 2023-05-05:
Not sure what to suggest, except to make sure you are absolutely using the correct files in both cases. There can be no substitutions - each is unique unlike previous releases.

SensorsIot, 2023-05-12:
I tied it with the files you provided, and it still does not work. Is there a way to see if the upgrade worked or if I still have the old files on the file system? In which file is the configuration of the radios/basic screen stored so that I can check if this file contains the proper information?
I tried the upgrade also with the command prompt, but also there, I did not see if the upgrade worked because it immediately disconnected the SSH connection after issuing the sysupgrade command.
In the next few days, I should get a new router to try. But of course, it would be great if I could debrick the old one...