aredn/aredn #820: AREDN Release 3.23.4.0 not compatible with Bullet M2Ti

convolved-KY4G, 2023-05-04:
A buddy has two Bullet M2Ti nodes.  They were running fine with AREDN Release 3.22.12.0, but upon upgrading them to 3.23.4.0 they did not function.  They each required multiple "factory resets," but eventually were able to be restored to 3.22.12.0 without having to install the OEM firmware.  Based on this, I would recommend changing the stability on the "Supported Devices" page to unstable, not recommended, or something like that.  I hesitate to use the word "brick" because with enough attempts, they were both recovered and rolled back.  Perhaps they should be "un-released" until the problem has been solved.

Stephen - KY4G

aanon4, 2023-05-04:
The Bullet M2Ti is not listed as compatible in the supported devices file already.

convolved-KY4G, 2023-05-04:
Ah, I see I put the wrong model in originally.  I have now corrected the title.  It was a Bullet M2 XW that he had the problem with.  It still shows unknown stability and released status.
[image]

aanon4, 2023-05-04:
Unfortunately I dont have one of these devices. If your buddy wants to send me one I can get it working. I'll send it back once I'm done.

convolved-KY4G, 2024-08-21:
Update: I checked with the folks that owned the bullet M2 XW devices and at least two of them have accepted firmware version 3.24.6.0 and appear to be working fine.  Given that they seem to be working now, I am closing this issue as resolved.

convolved-KY4G, 2024-08-21:
Since some testing has been done with 3.24.6.0, it may be that the supported hardware list may need to reflect that.

closed 2024-08-21 by convolved-KY4G