aredn/aredn #821: rc.local depreciated

KenArck, 2023-05-04:
I notice that ethtool is being used to set eth0 to non auto negotiate and set speed to 100 Mbs. 

This command does not work at boot as rc.local was depreciated

aanon4, 2023-05-04:
I see that it's being run on 3.23.4.0 - are you not seeing this?

KenArck, 2023-05-04:
Nope. Have to manually run it after the node boots in order to actually set eth0

aanon4, 2023-05-04:
Could you add the line
```
touch /tmp/rc-local-ran
```
At the top of /etc/rc.local and then reboot. I did this and the file was created after rebooting.

KenArck, 2023-05-04:
Nope

`root@WA7ABU-PB400-AC-REV:~# ls /tmp
TZ                   olsrd.watchdog       shm
dhcp.leases          overlay              snr.dat
etc                  resolv.conf          snrlog
lib                  resolv.conf.auto     state
lock                 resolv.conf.d        station_monitor.log
log                  rssi.dat             sysinfo
lqm.info             rssi.log             tmp
manager.log          rssi_ath10k.log      vtun
node.history         run                  web
root@WA7ABU-PB400-AC-REV:~#`

KenArck, 2023-05-04:
Right, it's a Powerbeam AC5 400

file is attached (just downloaded it prior to posting)
[supportdata-WA7ABU-PB400-AC-REV-202305041312.gz](https://github.com/aredn/aredn/files/11401165/supportdata-WA7ABU-PB400-AC-REV-202305041312.gz)

aanon4, 2023-05-04:
I see this in your log file:
```
Thu May  4 12:27:33 2023 daemon.notice procd: /etc/rc.d/S95done: /etc/rc.local: line 4: itouch: not found
```
Which suggests /etc/rc.local was being run as expected.

KenArck, 2023-05-04:
interesting.  So the issue must be rc.local is running BEFORE eth0 comes up?

aanon4, 2023-05-04:
I'm not sure why it would do anything on this hardware anyway as it should not have a matching board id to trigger the ethtool command.

aanon4, 2023-05-04:
Interestingly though, it looks like we might be able to remove this completely, as OpenWRT has fixed the driver issue which required this on some hardware.

KenArck, 2023-05-04:
The problem is it is connected to a Cisco router than is a #$@#$*$# to work with and won't auto negotiate properly. Unfortunately, the node is in a difficult to reach site so we're kinda stuck with the Cisco router for now. 

When we manually run ethtool, the router becomes happy and we assumed the rc.local call should do that for us (it is there by default, no?)

KenArck, 2023-05-04:
I just caught something

"itouch"????? Let me check I didn't fudge the entry in rc.local

aanon4, 2023-05-04:
Yes ... but as you note, it doesnt have the desired effect on eth0. The networking has changed substantially in the recent release, and clearly this code is no longer having the desired effect. I suspect, for our purposes at least, it's no longer required as the driver has been fixed.

Which doesnt help your situation much.  I'm not sure were best to put code which is guaranteed to run once the network configuration has been completed.

aanon4, 2023-05-04:
I notice in the logs that the node appears to have successfully negotiated 100Mb full duplex with your switch. No idea what your switch thinks of course.

KenArck, 2023-05-04:
Yes, switch not router. In any case, we have major route flopping through that node until we run ethtool after a reboot. Then it settles down

aanon4, 2023-05-04:
I should note that any changes you make to /etc/rc.local will not survive an upgrade.

KenArck, 2023-05-04:
Understand

aanon4, 2023-05-04:
If you put your ethtool at the end of /etc/rc.d/S99local it should happen very later in the boot process.

KenArck, 2023-05-04:
Worth a try, thanks!

aanon4, 2023-05-04:
Closing. /etc/rc.local is still executed and has not been deprecated. It appears to be executed earlier in the boot process than before.

closed 2023-05-04 by aanon4