aredn/aredn #822: Nanobridge M3 not working after update to latest stable release

ai6bx, 2023-05-05:
I have successfully updated the majority of my nodes. I started updating a Nanobridge M3 that was successfully running one of the pre-release lite versions only to have it not reboot. I have tried accessing via 192.168.1.1 thinking it might have gone back to first boot. No luck. I have tried rebooting several times with no changes. I have run Angry IP Scanner and only see my laptop and 192.168.1.20, the tough switch.

I have tried with a power brick with the same results. The node is not easy to get to so am looking for other potential remedies before renting an extension ladder tall enough to get to my son’s roof. Is it possible to get it back to TFtp mode and resend the update?

thank you,

Keith AI6BX

aanon4, 2023-05-05:
I have a nanobridge m3 myself, and the release certainly runs on it, although it's always a problematic upgrade as it has so little free memory.

I dont know if these general Ubiquti steps work on non-Rocket radios, but they might be worth trying?
```
Unplug the Ubiquiti radio from the POE Adapter

Push a pin into the POE Adapter reset button (on the model I have, the reset button was on the bottom, underneath the mounting plate)

While holding the reset button, plug the Ubiquiti radio into the POE adapter and continue to hold the reset button for 45 seconds

The Ubiquiti radio is now ready for the AREDN factory firmware update. It's listening on 192.168.1.20

TFTP the factory firmware to the Ubiquiti radio and you're back in business
```

ai6bx, 2023-05-05:
I will have to see if I still have a brick with the reset button.Sent from my iPad