aredn/aredn #825: GL-INET AR750 (Creta) Invalid Wi-Fi Channels for 2GHz MESH

k1ky, 2023-05-06:
Nightly 20230506 and a few previous? Invalid Wi-Fi 2GHz selection channels show 5GHz channels instead of 2.4GHz options.
C:\Users\k1kyt\My Drive\MESH_HAMNET\FIRMWARE\SCREENSHOTS\2023-05-06 16_19_49-K1KY-AR750-750-P1 setup.png
[supportdata-K1KY-AR750-750-P1-202304292228.gz](https://github.com/aredn/aredn/files/11413421/supportdata-K1KY-AR750-750-P1-202304292228.gz)

k1ky, 2023-05-06:
OK - Actually I see one issue already. My LAN Access Point is using 2GHz - should have been 5GHz, so additional 2GHz MESH sould not be allowed if already being used for Wi-Fi LAN Access Point.  The LAN Access Point (was) using 5GHz channels, not sure what changed there.  I'll try to change it and see what happens.

k1ky, 2023-05-06:
The plot thickens.  The LAN Access Point was using 5GHz channel 36, but when I checked the 2GHz MESH RF box, the LAN Access channel changed to 2GHz, so the real issue is inability to choose 2GHz MESH and 5GHz Lan Access point.

aanon4, 2023-05-06:
Could you tell me the configuration of the mesh and the AP before you did the upgrade? I'd like to setup things on my AR750, do the upgrade, and see what's messed up in the internal state.

aanon4, 2023-05-06:
Well even starting from a sensible place, it can get confused very quickly ... will take a look at this logic (which needed work anyway)

k1ky, 2023-05-06:
Definitely some weird issues here.  Should not allow both Wi-Fi Lan and MESH to use 5GHz. 
[image]

k1ky, 2023-05-06:
Before the upgrade, MESH RF was turned off. Lan Access Point was on 5GHz Channel 36. WAN Wi-Fi Client was previously being used on 2GHz, but that had been disabled at the time of the upgrade.

k1ky, 2023-05-06:
Same issue(s) with the Mikrotik HAP ac Lite
[image]

aanon4, 2023-05-08:
This should fix the problem.