aredn/aredn #833: Ubiquiti NanoBridge M5

K6AH, 2023-05-11:
This comes from LA9UI from his original email in Norwegian.  Here in Tromsø, we have a large number of Ubiquiti NanoBridge M5. They have behaved well until I started upgrading to 3.23.4.0. About 1 in 4 fail during the upgrade. Everything looks normal
the upgrade is complete. The red light stays on for a couple of minutes (connect), then it starts flashing,
2 seconds off and 2 seconds on. I can't connect to the radio. A radio out in the net has failed, they
others have been upgraded at home. I now have at least 6 lying around that have failed; all show the same
symptoms. Is this something that others have also experienced?
I've tried searching the Forum, but I don't see any way to filter the search.
On versions 3.22.12.0 and earlier bullet-m was the correct image, but I see that now it is
nanobridge-m that applies. A couple of the first ones were upgraded with bullet-m and it seemed to work.
What are the differences bullet-m, bullet-m-ar7241 and nanobridge-m?
73, de Hans-Petter, LA9UI

k1ky, 2023-05-11:
I just did a successful Nanobridge-m firmware upgrade on my M5 unit (just upgraded from Nightly 2526 to Nightly 20230510).  Both were already the tiny Nanobridge versions.  The problem may be bridging the different model versions, which I experienced before with Powerbeam units. I'll check to see if I still have any NBM5 units running bullet firmware.

k1ky, 2023-05-12:
My 2nd upgrade   was successful from Nightly 21xx Nanobridge  to tiny 20230510 Nanobridge without any problems, so all I can report is no issues so far. We also have a units successfully running 3.23.4.0, N-2432 and another 20230510 in active service.

aanon4, 2023-05-13:
@K6AH There's a fair amount of information if would be good to get. As some work and some dont, it would be useful to see if LA9UI can determine anything particular about the failing ones that might be common. The red light on then flashing .. that's just normal behaviour, with the final flashing red light indicating that the node has not managed to connect to any other nodes. The fact that it's flashing like that means it is running AREDN. So it would be good to get better information about what "failed" means in this context. No longer available on RF? Doesnt connect to ethernet? 

Regarding the different binary names, it would be good to get some idea of what was upgraded with what, what worked and what didnt. 

I dont have a NanoBridge M5 but I do have a NanoBridge M3. It has very little memory and is a pain to upgrade and often requires some power cycling. I assume this much has been tried when connected to ethernet?