aredn/aredn #837: Time and date not working since 3.23.4 release

ai6bx, 2023-05-13:
Good morning:

In looking at a number of my nodes I noticed the date and time under setup has not changed since the date and time I updated the firmware. On some I have been able to get it going by hitting the apply settings button, save, and reboot. This is not a consistent fix however.

Keith - AI6BX

ai6bx, 2023-05-14:
This is likely linked to ticket 838 which I just closed by mistake. The problem with dropping links an rebooting is not resolved and I now note that upon reboot, the system time and date again revert back to the time and date I applied the 3.23.4 production release. I am attaching the diagnostic
[supportdata-AI6BX-1-RM3-East-202304272232.gz](https://github.com/aredn/aredn/files/11472340/supportdata-AI6BX-1-RM3-East-202304272232.gz)
 file as the node just rebooted again two hours ago.

aanon4, 2023-05-14:
The log file claims the node cannot contact any NTP servers. Is this expected on your mesh?

aanon4, 2023-05-16:
I had a quick look at a random sample of M3 nodes in the SoCal network running 3.23.4.0, and all their times were correct.

aanon4, 2023-05-30:
Any update on this? The problem seems to be on this specific node, yes?

ab7pa, 2023-07-19:
Unable to reproduce, so closing for now. Can reopen if needed.