aredn/aredn #838: Rocket M3 intermittently dropping RF link and rebooting since 3.23.4 update

ai6bx, 2023-05-13:
The title says it all. There is no consistency to the time intervals between this happening. What is consistent is the RF drops happen multiple times at varied intervals before a power cycle reboot occurs.

Keith - AI6BX

Orv, 2023-05-13:
I don't see reboots, but I do see a couple of short strong 5 GHz links dropping periodically, when they never should.  Unsure how to troubleshoot.

ai6bx, 2023-05-14:
Perhaps this will help identify the issue.
[supportdata-AI6BX-1-RM3-East-202304272232.gz](https://github.com/aredn/aredn/files/11472331/supportdata-AI6BX-1-RM3-East-202304272232.gz)

ai6bx, 2023-05-14:
The node just spontaneously rebooted again 16 minutes ago. Is there data I can collect that will help identify issues?

aanon4, 2023-05-16:
Not really - the nodes dont have the ability to log why they reboot. Is there any possibility there's a power issue with this node? Are you having any trouble with your other Rocket M3 hardware?

k1ky, 2023-05-18:
Suggest measuring voltage at the node if possible?  How much cable and verify 24v 1a power injector?

ai6bx, 2023-06-05:
It looks like the Rocket radio is failing so will plan a trip to replace. @Orv, did you figure out how to troubleshoot the 5ghz drops? I am having some of the same with evidence manifesting in linked repeater and packet drops on ping.

Orv, 2023-06-05:
Keith, nightly build 20230603 resolved the specific instances of link 
dropouts I was seeing.  Recommend updating some or all of your 
problematic links (especially those using 802.11n gear) to the latest 
nightly build and monitoring their behavior.

ai6bx, 2023-06-05:
Thanks. I will look at that and update as soon as I can access.Sent from my iPad

ai6bx, 2023-06-06:
Loaded last night and monitoring.Sent from my iPad