aredn/aredn #842: UBNT Nanobeam AC Gen1 - Unspported Device

moordere, 2023-05-17:
node: 
model: 0xe9f5

!!!! UNSUPPORTED DEVICE !!!!
boardid: 0xe9f5
Device HAS BEEN TESTED AS UNSUPPORTED

br-lan 80:2A:A8:F9:8E:7A
eth0   80:2A:A8:F9:8E:7A
eth0.1 80:2A:A8:F9:8E:7A
eth0.2 80:2A:A8:F9:8E:7A
tunl0  00-00-00-00-00-00-00-44-00-00-00-00-00-00-00-00
wlan0  80:2A:A8:F8:8E:7A

/proc/cpuinfo
system type		: Atheros AR9342 rev 2
machine			: Ubiquiti NanoBeam AC Gen1 (WA)
processor		: 0
cpu model		: MIPS 74Kc V4.12
BogoMIPS		: 266.64
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 32
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16 dsp dsp2
Options implemented	: tlb 4kex 4k_cache prefetch mcheck ejtag llsc dc_aliases perf_cntr_intr_bit perf
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

nvram
hsmmmesh.settings=settings

aanon4, 2023-05-17:
Thanks. I'll look at adding this. Which image did you use?

moordere, 2023-05-17:
http://downloads.arednmesh.org/afs/www/?version=3.23.4.0&target=ath79%2Fgeneric&id=ubnt_nanobeam-ac

moordere, 2023-05-22:
More information this is a NBE-5AC-16 (Nanobeam AC 5GHz 16 dBi) It uses the WA board