aredn/aredn #846: Ubiquiti Rocket 5AC - Unsupported Device

moordere, 2023-05-22:
node: 
model: 0xe3f5

!!!! UNSUPPORTED DEVICE !!!!
boardid: 0xe3f5
Device HAS BEEN TESTED AS UNSUPPORTED

br-lan 44:D9:E7:2B:0A:92
eth0   44:D9:E7:2B:0A:92
eth0.1 44:D9:E7:2B:0A:92
eth0.2 44:D9:E7:2B:0A:92
tunl0  00-00-00-00-00-00-00-44-00-00-00-00-00-00-00-00
wlan0  44:D9:E7:2A:0A:92

/proc/cpuinfo
system type		: Qualcomm Atheros QCA9558 ver 1 rev 0
machine			: Ubiquiti Rocket 5AC Lite     #####Device is non Lite version R5AC-PTMP#####
processor		: 0
cpu model		: MIPS 74Kc V5.0
BogoMIPS		: 358.80
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 32
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16 dsp dsp2
Options implemented	: tlb 4kex 4k_cache prefetch mcheck ejtag llsc dc_aliases perf_cntr_intr_bit cdmm contextconfig perf
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

nvram
hsmmmesh.settings=settings

moordere, 2023-05-22:
Board type is XC

aanon4, 2023-05-24:
Do you have a link to this device? I've never heard of it but someone else mentioned it to me today.

moordere, 2023-05-24:
https://dl.ui.com/qsg/R5AC-PTMP/R5AC-PTMP_EN.html
https://community.ui.com/questions/Difference-between-R5AC-Lite-and-R5AC-PTMP/ef9d8ddd-739c-49ac-b167-8561d2bcf60c

n9jim, 2023-09-12:
i have 4 ac-ptmp's... just loaded one from stock to 3.23.8.0
it displays with red bar in main screen: Unsupported device

 I note that the PTMP received signal strength appears to be 20 dB less than the other direction to a 5AC-lite, and the "noise level" is also lower. 
AC lite sees AC ptmp as -58dBm with -113 noise 53dB S/N on chart.
AC ptmp sees AC lite as -80 -120 40dB S/N

Jim
N9JIM

aanon4, 2023-11-15:
This is still not supported, even in the latest OpenWRT 23.05.0 release. Do we know how well it performs using the AC Lite build? Maybe the numbers reported are not a good reflection of how well it works?

n9jim, 2023-11-17:
TimI loaded the October nightly today.  Behavior is the same. Rx is about 20dB lower in actual performance, and reported.JimSent from my iPhone

aanon4, 2023-11-17:
Just to clarify. If you do a bandwidth tests it performs about as poorly as you'd expect given the lower SNR?

n9jim, 2023-11-17:
Communication appears to be “oneway”.Signal levels can be exchanged, but not a good enough connection to see the Home Screen when using the ptmp device.M5 reports the ptmp is 46 db snr on neighbor screen. Chart reports 48-50 db (-62 signal), tx rate 6.5mbps mcs n/a , rx 26mbps, mcs3Ptmp reports the m5 as 30dB snr on the neighbor screen.  Chart reports signal as -85 noise -120, 35 snr, tx19.5mbps mcs2, rx 6.5mbps mcs naSent from my iPhone

n9jim, 2023-11-17:
From a more direct path & nsm5Sent from my iPhone