aredn/aredn #848: Enhancement: allow disabling WAN access to HTTP admin

bwarden, 2023-05-24:
For nodes that are connected to a WAN that's not controlled by the node owner, or worse, on a public IP address, it's desirable to be able to completely disable HTTP access to even the node status pages from other WAN users.

Could we add an option to block port 8080 from the WAN?

Prototype:
```bash
/lib/functions.sh

SECTION=firewall

remove_rule() {
  local config="$1"
  local find_src="$2"
  local find_dest_port="$3"

  config_get src "${config}" src
  config_get dest_port "${config}" dest_port

  if [[ "${src}" == "${find_src}" ]] && [[ "${dest_port}" == "${find_dest_port}" ]]; then
    uci show ${SECTION}.${config}
    echo "uci -q delete ${SECTION}.${config}"
  fi

}

config_load ${SECTION}
config_foreach remove_rule rule wan 8080
```

AI7NC, 2023-05-26:
Would would need port 80 removal as well correct? Seems I can access via both 80 and 8080 over WAN interface.

aanon4, 2023-05-26:
I'll be honest, I'm not sure why we'd want any sort of incoming access on the WAN port and I'm a little horrified to discover that we do. I didnt think this was the case and I wonder if I messed something up in the latest release (we had to rewrite the entire firewall stack because of underlying changes to OpenWRT).

So I'm inclined to not ever make this an option, but just to block all incoming traffic from WAN (except tunnels). Unless anyone thinks there's a good reason not to do that?

AI7NC, 2023-05-26:
As an FYI, N7IME also mentioned that port 2222 seems to be open on WAN for SSH. It makes sense for this to be an option for those that configure a SSH key but maybe shouldn't be on by default.

AI7NC, 2023-05-26:
I can't think of a strong reason to need access. If you block it by default in the firewall rules, can you manually re-enable access by creating a port forward on 80 or 8080 to localhost to re-enable it? 

[image]

mathisono, 2023-05-26:
I have to say that I do visit nodes http via the WAN.

bwarden, 2023-05-26:
> I'll be honest, I'm not sure why we'd want any sort of incoming access on the WAN port and I'm a little horrified to discover that we do. I didnt think this was the case and I wonder if I messed something up in the latest release

It's been enabled in AREDN firmware for years, so it's not your fault :) I think it was even in HSMM.

For me it's handy on my home network, where I have control of it, but obviously in other deployments it's undesirable. I'd like options to enable SSH and 80/8080 access on the WAN interface, even if they're off by default.

For what it's worth, OpenWRT manages SSH by running separate dropbear instances on the WAN and LAN interfaces, as configured by luci. I think we're running both dropbear and uhttpd on 0.0.0.0.

We could even get by with manual config in /etc/config/firewall and /etc/config/dropbear, as long as the AREDN web interface doesn't clobber those entries. But two toggles in advanced config would be super handy.

aanon4, 2023-05-29:
https://github.com/aredn/aredn/pull/854

AI7NC, 2023-06-02:
Thanks for moving fast on this Tim!

bwarden, 2023-07-13:
Looks good

closed 2023-07-13 by bwarden