aredn/aredn #851: NB 20230526-158773f doesn't display properly on ac2

Orv, 2023-05-26:
[image]
[supportdata-W6BI-SHACK-NODE-202305262059.gz](https://github.com/aredn/aredn/files/11579887/supportdata-W6BI-SHACK-NODE-202305262059.gz)

hAP ac2 with NB 20230526-158773f doesn't display Basic Setup page as expected.  Also gets Bad Gateway when attempting to save that page.
Attached:  Screenshot & support data file

aanon4, 2023-05-28:
@Orv Quick question - was the wifi disabled on this device?

Orv, 2023-05-28:
It was.

aanon4, 2023-05-28:
Fix is in. Give the nightly a whirl tomorrow.

Orv, 2023-05-28:
Roger wilco.

Orv, 2023-05-28:
Nightly Build 20230528 resolves this issue - the Setup page now displays as expected.

closed 2023-05-28 by Orv